arduino/Arduino #5450: arduino_debug --port option broken in 1.6.12

SB618, 2016-10-04:
Calling arduino_debug.exe from command line using a command like the following crashes with an exception parsing the serial port.  This does not crash in 1.6.11 and previous.

arduino_debug --port COM1 --board arduino:avr:uno --upload "C:\Program Files (x86)\Arduino\libraries\SD\examples\Files\Files.ino"

Loading configuration...
java.lang.NullPointerException
        at processing.app.BaseNoGui.selectSerialPort(BaseNoGui.java:1088)
        at processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.guardedMain(Base.java:219)
        at processing.app.Base.main(Base.java:132)

kevans91, 2016-10-04:
Hi,

Related: #5413

SB618, 2016-10-04:
Sorry, did not see this issue already open.  Thanks.

closed 2016-10-04 by SB618