arduino/Arduino #5454: IDE 1.6.12

matyshao, 2016-10-05:
If I click "upload" button in Ardublock will error.

Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Editor.setText(Ljava/lang/String;)V
 at com.ardublock.ArduBlockTool.didGenerate(ArduBlockTool.java:72)
 at com.ardublock.core.Context.didGenerate(Context.java:271)
 at com.ardublock.ui.listener.GenerateCodeButtonListener.actionPerformed(GenerateCodeButtonListener.java:171)
 at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
 at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
 at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
 at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
 at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
 at java.awt.Component.processMouseEvent(Component.java:6533)
 at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
 at java.awt.Component.processEvent(Component.java:6298)
 at java.awt.Container.processEvent(Container.java:2236)
 at java.awt.Component.dispatchEventImpl(Component.java:4889)
 at java.awt.Container.dispatchEventImpl(Container.java:2294)
 at java.awt.Component.dispatchEvent(Component.java:4711)
 at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4888)
 at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4525)
 at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4466)
 at java.awt.Container.dispatchEventImpl(Container.java:2280)
 at java.awt.Window.dispatchEventImpl(Window.java:2746)
 at java.awt.Component.dispatchEvent(Component.java:4711)
 at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
 at java.awt.EventQueue.access$500(EventQueue.java:97)
 at java.awt.EventQueue$3.run(EventQueue.java:709)
 at java.awt.EventQueue$3.run(EventQueue.java:703)
 at java.security.AccessController.doPrivileged(Native Method)
 at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
 at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
 at java.awt.EventQueue$4.run(EventQueue.java:731)
 at java.awt.EventQueue$4.run(EventQueue.java:729)
 at java.security.AccessController.doPrivileged(Native Method)
 at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
 at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
 at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
 at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
 at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
 at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
 at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
 at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

facchinm, 2016-10-05:
The editor has been refactored on 1.6.12 and we tried to tell every extension developer to update their tools, but we probably missed Ardublock. Will contact them asap so they can release an updated version. Thanks for reporting!

facchinm, 2016-10-05:
Reported here https://github.com/taweili/ardublock/issues/159

birdbraintech, 2016-10-25:
Thanks facchinm for the fix! We tested it on our [branch of Ardublock](https://github.com/BirdBrainTechnologies/HummingbirdArdublock) and it now works in 1.6.12 and appears backwards compatible with all 1.6 versions of the Arduino IDE that we've tested.

closed 2016-10-25 by facchinm

margamasse, 2016-11-07:
thanks birdbraintech

I tested our ardublock-master and it works fine but i have an error in the file OpenBlocksFrame.java and the 
save image button doesn't work.

		JButton saveImageButton 
.....

				
				workspace.getBlockCanvas().getPageAt(0).getJComponent().paint(g);

......

"the method getPageAt(int) from the type BlockCanvas is invisible"

birdbraintech, 2016-11-08:
I am not getting that error (Windows 10, java 8) but our Ardublock branched from the main branch two years ago and not all of the Ardublock changes from 2015 have been integrated, so it could be that the behavior is different.

rossp-fll, 2017-01-06:
Still seeing this error in 1.6.13, Win 10, Java 8-U111, AMD A4.
Are there any known issues with AMD?