arduino/Arduino #5456: Arduino IDE 1.6.10/1.6.12 Command Line Issue

PROGRAMINO, 2016-10-05:
The Command Line have a problem with Arduino NANO and upload.

Sample for 1.6.10:
--------------------
java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": CreateProcess error=2, Das System kann die angegebene Datei nicht finden
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:130)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:209)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1187)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2409)
	at processing.app.Base.<init>(Base.java:436)
	at processing.app.Base.guardedMain(Base.java:223)
	at processing.app.Base.main(Base.java:136)
Caused by: java.io.IOException: CreateProcess error=2, Das System kann die angegebene Datei nicht finden
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more

Sample for 1.6.12:
--------------------
"--board arduino:avr:nano:cpu=atmega328 --port COM55 --upload " & """" & path & """"

bring the following error...

java.lang.NullPointerException
	at processing.app.BaseNoGui.selectSerialPort(BaseNoGui.java:1088)
	at processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)
	at processing.app.Base.<init>(Base.java:273)
	at processing.app.Base.guardedMain(Base.java:219)
	at processing.app.Base.main(Base.java:132)

Arduino 1.6.11/1.6.9 and lower version works fine!

per1234, 2016-10-06:
Please test with the hourly build:
http://www.arduino.cc/en/Main/Software#hourly
The problem you encountered with Arduino IDE 1.6.12 should be fixed.

You should always test with the hourly build before reporting an issue to make sure that the issue hasn't already been fixed.

facchinm, 2016-10-06:
Thanks @per1234 , indeed it was fixed in 0a8e7a14f149534076455d9dd5fe17c44285686c so latest nightly is ok

closed 2016-10-06 by facchinm

PROGRAMINO, 2016-10-06:
Works fine! 
Good job!