arduino/Arduino #5461: Add -fshort-enums

NicoHood, 2016-10-06:
It took me an extremely long time to debug a c code that used an uint8_t enum (that requires the 8 bit and not 16 bit size). The reason is that `-fshort-enums` is not used for arduino. We should add this at least to the c flags, as you cannot define the enum size as in c++.

xcvista, 2016-10-06:
Is there any other way to do this? I remember that C++11 supports explicit definition of enum width. LLVM/clang have long added this C++ feature as an extension to C already, and I wonder if GCC followed suit.

NicoHood, 2016-10-06:
Yes c++ can do something like `enum myname : uint8_t` but not for c files. In c you can only use the compiler switch. Also in general its a good idea to short enums to th minimum size, as this saves space.

matthijskooijman, 2016-10-06:
@xcvista, C++11 enum classes support this, but they are significantly different from normal enums in other ways too, so I think adding this option is a good idea. The primary downside (and why this is not enabled by default) of this option is reduced binary compatibility, but since Arduino always compiles everything from source, this should not really be a problem.

oqibidipo, 2016-10-06:
You can use `__attribute__((packed))`

https://gcc.gnu.org/onlinedocs/gcc-4.8.5/gcc/Type-Attributes.html
> `packed`

> This attribute, attached to struct or union type definition, specifies that each member (other than zero-width bit-fields) of the structure or union is placed to minimize the memory required. When attached to an enum definition, it indicates that the smallest integral type should be used.
Specifying this attribute for struct and union types is equivalent to specifying the packed attribute on each of the structure or union members. Specifying the -fshort-enums flag on the line is equivalent to specifying the packed attribute on all enum definitions.

> In the following example struct my_packed_struct's members are packed closely together, but the internal layout of its s member is not packed—to do that, struct my_unpacked_struct needs to be packed too.

>          struct my_unpacked_struct
           {
              char c;
              int i;
           };
          
>          struct __attribute__ ((__packed__)) my_packed_struct
            {
               char c;
               int  i;
               struct my_unpacked_struct s;
            };
> You may only specify this attribute on the definition of an enum, struct or union, not on a typedef that does not also define the enumerated type, structure or union.

NicoHood, 2016-10-06:
Thanks! This is a reasonable solution that works without the compiler switch. However we should add it anyways.

```c
typedef enum __attribute__((packed)) {
    FLASH_OPCODE_READ_LOW_POWER = 0x01,
    FLASH_OPCODE_READ_MODIFY_WRITE_BUF1 = 0x58,
    FLASH_OPCODE_READ_INTO_BUF2 = 0x55,
} flash_opcode_t;
```

xcvista, 2016-10-06:
Or maybe add a few new Preferences item (`CFLAGS`, `CPPFLAGS`, `CXXFLAGS` and `LDFLAGS`) so the user can add any arbitrary flags to the compiler command line. `CFLAGS` for C, `CXXFLAGS` for C++, `CPPFLAGS` for C, C++ and assembler, and `LDFLAGS` for linking.