arduino/Arduino #5462: New library - SixAxisRing

Patrick-Thomas, 2016-10-06:
Could you add my library to your database please?
https://github.com/Patrick-Thomas/SixAxisRing

It's tagged as a pre-release at the moment, hope this OK

agdl, 2016-10-07:
@Patrick-Thomas actually I suggest you to put the examples folder showing how to use this library and then adding it to the library manager

Patrick-Thomas, 2016-10-11:
Hi Arturo, I've added two examples as per your request. Let me know if this is sufficient. Patrick

cmaglie, 2016-10-18:
The library has been added to the library manager.

BTW, the examples are still not published because you need to make another release.
Please follow this steps (in this exact order to avoid errors):

1) change the version in `library.properties` to `0.0.2`
2) commit and push
3) create a new tag `0.0.2`

Once done our server should pick up the changes automatically within an hour without further actions.

closed 2016-10-18 by cmaglie

Patrick-Thomas, 2017-01-13:
Hi, I tried installing the library from scratch. Although the examples are downloaded correctly along with the .h and .cpp files, they are not visible under File->Examples.

How can I make the examples visible here?
Thanks