arduino/Arduino #5463: Arduino fails to compile a sketch with tabs on ubuntu if path contains spaces

Vincent14, 2016-10-06:
Tested on Ubuntu with Arduino 1.6.12 taken from arduino.cc website.

facchinm, 2016-10-07:
Hi @Vincent14 , the bug is not confirmed (I could not reproduce on my systems). Could you post the complete verbose compilation output and a minimal multitab sketch triggering the issue?

Vincent14, 2016-10-07:
You're right, the problem is not the space but the accent "à". Have a try with [this project](http://dl.free.fr/grhM2Llve). Here is the full trace:

> /media/Stockage/Bureau/à/cubeled_v1.2/cubeled_v1.2.ino: In function 'void setup()':
> cubeled_v1.2:28: error: 'cubeBegin' was not declared in this scope
>    cubeBegin();
>              ^
> cubeled_v1.2:29: error: 'initLeds' was not declared in this scope
>    initLeds();
>             ^
> cubeled_v1.2:30: error: 'loadParams' was not declared in this scope
>    loadParams();
>               ^
> /media/Stockage/Bureau/à/cubeled_v1.2/cubeled_v1.2.ino: In function 'void loop()':
> cubeled_v1.2:39: error: 'refreshFrame' was not declared in this scope
>    if (refresh && changed) refreshFrame();
>                                         ^
> cubeled_v1.2:44: error: 'drawLeds' was not declared in this scope
>        drawLeds();
>                 ^
> cubeled_v1.2:48: error: 'drawLeds' was not declared in this scope
>    else drawLeds();
>                  ^
> /media/Stockage/Bureau/à/cubeled_v1.2/cubeled_v1.2.ino: In function 'void serialEvent()':
> cubeled_v1.2:58: error: 'parse' was not declared in this scope
>      parse(input);
>                 ^
> /media/Stockage/Bureau/à/cubeled_v1.2/config.ino: In function 'void refreshFrame()':
> config:73: error: 'copyFrame' was not declared in this scope
>    copyFrame(bufferFrame, actualFrame);
>                                      ^
> /media/Stockage/Bureau/à/cubeled_v1.2/parser.ino: In function 'void parse(byte)':
> parser:54: error: 'execute' was not declared in this scope
>          if (input == EOP) execute();
>                                    ^
> /media/Stockage/Bureau/à/cubeled_v1.2/parser.ino: In function 'void execute()':
> parser:265: error: 'MOVE' was not declared in this scope
>          MOVE(buffer[0], 1, false);
>                                  ^
> parser:269: error: 'MOVE' was not declared in this scope
>          MOVE(buffer[0], buffer[1], false);
>                                          ^
> parser:276: error: 'MOVE' was not declared in this scope
>          MOVE(buffer[0], 1, true);
>                                 ^
> parser:280: error: 'MOVE' was not declared in this scope
>          MOVE(buffer[0], buffer[1], true);
>                                         ^
> exit status 1
> 'cubeBegin' was not declared in this scope

oqibidipo, 2016-10-08:
This looks like #5216 and #5277.

Vincent14, 2016-10-08:
Yes, you're right

dlabun, 2016-10-13:
Any non-ASCII character will give you problems... There an open PR for Arduino builder that should fix the issue. [https://github.com/arduino/arduino-builder/pull/178]

facchinm, 2016-10-20:
Closing as duplicate of #5216

closed 2016-10-20 by facchinm