arduino/Arduino #5466: Rotary encoder page on Arduino playground website is a mess

clovisf, 2016-10-08:
Today I entered for the first time in the [rotary encoders](http://playground.arduino.cc/Main/RotaryEncoders) page at arduino Playground ( http://playground.arduino.cc/Main/RotaryEncoders ). I found one minor issue and one major, that I would like to see corrected (or even help correct).

- **Minor issue**: this is the longest (content-wise) Playground page I know, and the problem is that the user (myself) gets overwhelmed with that much information to scavenge through. There could be implemented a different approach on the matter of which content is really "important" to be there.

- **Major issue**: there are two handfuls of examples on how to use a rotary encoder, every and each one very similar to each other; most of them use the concept of interrupts on code, and what changes between them is basically the number of interrupts (or pins) used (from 1 to 2). T**here is even a childish conflict of interests** between the library contributed by one and another member, where one attacks the other on the reasons his library is better/faster. I suggest shrinking it down to one or two main examples (one without and one with interrupts for example), as well as delimiting or better informing the models and types of rotary encoders to the most widely or cheaply available to the general public.

agdl, 2016-10-10:
@clovisf thank you for the interest in improving this! Actually playground is a wiki and everyone with an Arduino website ID can contribute it!

per1234, 2017-07-05:
I did a lot of cleanup and organization on the page. Even though there is still the same amount of content it is now much easier to navigate and read. If anyone wants to make any improvements to that page or any other Playground page you are welcome to do so.

closed 2017-07-05 by per1234