arduino/Arduino #5469: New library version: KONNEKTING

tuxedo0801, 2016-10-10:
Hi, I updated my arduino library to a new version (https://github.com/KONNEKTING/KonnektingDeviceLibrary/releases/tag/1.0.0-BETA4) and now waiting for an update in arduino ide's library manager. I read on the wiki that the job runs every hour? Now it's about 1,5hrs. Nothing happend.

Is there any issue?

br,
 Alex

Chris--A, 2016-10-11:
From what I can see, the versioning is the issue.

Rather than using betaX just use semantic versioning.  `Major.Minor.Revision`.
Major: Rewrites, breaking changes.
Minor: New features.
Revision: Bug fixes and updates.

The lib manager is not picking up your previous releases or the next one because (guess) it sees the same version number (1.0.0) I do not think the beta part is playing a role here. It also is not useful as if you go to 1.0.1 after beta, there is no indication of how many revisions/increments have occured without scouring the history.

This could be fixed by:
- deleting the tags you currently have.
- using semantic versioning without betaX
- release a tag with the new version.

Then new versions will be recognised when you release a tag for them. (`1.0.0`, `1.0.1`, `1.1.0`, ...)

This is also stated here: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

> version - version of the library. Version should be semver compliant. 1.2.0 is correct; 1.2 is accepted; **r5, 003, 1.1c are invalid**

tuxedo0801, 2016-10-11:
Hey Chris,

the last time I added a version (1.0.0-beta3) it worked quite well?!

The FAQ (https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) says:

> A library is not valid when:
> ...
> its version number is not semver compliant
> ...

When I follow the link to the semver website (http://semver.org/), it says:

> Given a version number MAJOR.MINOR.PATCH, increment the:
> 
> MAJOR version when you make incompatible API changes,
> MINOR version when you add functionality in a backwards-compatible manner, and
> PATCH version when you make backwards-compatible bug fixes.
> **Additional labels for pre-release and build metadata are available as extensions to the MAJOR.MINOR.PATCH format**.

Because of that, my expectation was, that having kind of pre-release label is valid...mainly because of the fact, that 1.0.0-beta3 was detected and accepted.

Maybe I setup the github tag wrong. I did not set the checkbox "This is a pre-release". Maybe that makes a difference? I now set the checkbox and will see if this changes anything.

tuxedo0801, 2016-10-11:
I searched through the library json file and found another library, that uses a pre-release tag:

```
{
      "name": "ArduinoJson",
      "version": "5.1.0-beta.1",
      "author": "Benoit Blanchon \u003cblog.benoitblanchon.fr\u003e",
      "maintainer": "Benoit Blanchon \u003cblog.benoitblanchon.fr\u003e",
      "sentence": "An efficient and elegant JSON library for Arduino.",
      "paragraph": "Like this project? Please star it on GitHub!",
      "website": "https://github.com/bblanchon/ArduinoJson",
      "category": "Data Processing",
      "architectures": [
        "*"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/bblanchon/ArduinoJson-5.1.0-beta.1.zip",
      "archiveFileName": "ArduinoJson-5.1.0-beta.1.zip",
      "size": 1335057,
      "checksum": "SHA-256:de02e8a739b0b97b430f28dac51d675cc5796bf503446b9711f2a20c285a7eec"
    },
    {
      "name": "ArduinoJson",
      "version": "5.1.0-beta.2",
      "author": "Benoit Blanchon \u003cblog.benoitblanchon.fr\u003e",
      "maintainer": "Benoit Blanchon \u003cblog.benoitblanchon.fr\u003e",
      "sentence": "An efficient and elegant JSON library for Arduino.",
      "paragraph": "Like this project? Please star it on GitHub!",
      "website": "https://github.com/bblanchon/ArduinoJson",
      "category": "Data Processing",
      "architectures": [
        "*"
      ],
      "types": [
        "Contributed"
      ],
      "url": "http://downloads.arduino.cc/libraries/bblanchon/ArduinoJson-5.1.0-beta.2.zip",
      "archiveFileName": "ArduinoJson-5.1.0-beta.2.zip",
      "size": 1335021,
      "checksum": "SHA-256:2aba3015fc64f73f50cff3a4f31100037bea5a807ed0ae7d2ff79b028e7e6633"
    },
```

So using that kind of pre-release label is actually working. Question is: How?!

Chris--A, 2016-10-11:
Depending on how the parser is setup, the ArduinoJson tags might be working as there is a '.' between beta and the the number. But these are only guesses.
It'd be best to wait for a definitive answer from one of the team.

tuxedo0801, 2016-10-11:
****Waiting****

cmaglie, 2016-10-11:
Turned out it was a problem on the server, now it's fixed and the new version is published.

Sorry for the delay, and thanks for your patience!

closed 2016-10-11 by cmaglie