arduino/Arduino #5470: Pls. add new library: ZeroEEPROM

tuxedo0801, 2016-10-11:
https://github.com/KONNEKTING/ZeroEEPROM/releases/tag/1.0.0

Adds a kind of virtual EEPROM to ArduinoZERO/SAMD21G boards wit EEPROM.h like API.

cmaglie, 2016-10-14:
> Adds a kind of virtual EEPROM to ArduinoZERO/SAMD21G boards wit EEPROM.h like API

@tuxedo0801 
Following your thread here https://github.com/arduino/ArduinoCore-samd/issues/145, I have a proposal: what do you think if we add these files in the FlashStorage library and rename them to `FlashAsEEPROM.h`?

pros:
- the user doesn't need to install two libraries (ZeroEEPROM and FlashStorage)
- the include "FlashAsEEPROM.h" make it clear that is an emulated EEPROM and that the API may be slightly different, IMHO this may address the concerns expressed in https://github.com/arduino/ArduinoCore-samd/issues/145
- we could expand the library to support also SAM (Arduino Due) and other platforms as well

what do you think?

tuxedo0801, 2016-10-14:
I agree. That's a good idea. Will you add/rename the files and release a new version?

cmaglie, 2016-10-14:
Ok I'm on it

cmaglie, 2016-10-14:
here we go: https://github.com/cmaglie/FlashStorage/issues/7

closed 2016-10-19 by cmaglie