arduino/Arduino #5472: Missing circuit diagram

Johns46, 2016-10-11:
Not sure if this is the right place, but:

http://playground.arduino.cc/Code/ACPhaseControl  

The circuit diagram is missing.

per1234, 2016-10-12:
Duplicate of https://github.com/arduino/Arduino/issues/5129

I'm the user "pert" mentioned in #5129 but I only fixed some typos on that page, I didn't remove the diagram and have no connection with the creation of that page. The problem is the circuit diagram URL:
http://img27.imageshack.us/img27/6200/acphasecontrol.png
is dead. It's not available on the Wayback Machine. I did some searching on Google Images but none of the schematics match the part numbers specified on the Playground page.

I did find an image of that name in one of the Playground page author's GitHub repositories:
https://github.com/Ltalionis/lex-arduino-sketchbook/blob/master/AC%20Phase%20Control/AC_MC_testing_intchange/ac_control.png
However, the sketch folder it's in is a different sketch and the part number of the SSR doesn't match the one specified on the Playground page. I have sent a private message to the author, Italionis on the Arduino forum and will update the page if they provide a diagram or confirm the one I found is correct.

I have removed all references to the missing diagram from the Playground page to prevent future confusion. If you find or make a replacement diagram feel free to add it to that page or let me know and I'll do it.

Johns46, 2016-10-12:
Many thanks, have a good day.

 

John S.

 

http://img27.imageshack.us/img27/6200/acphasecontrol.png

https://github.com/Ltalionis/lex-arduino-sketchbook/blob/master/AC%20Phase%20Control/AC_MC_testing_intchange/ac_control.png

closed 2016-10-12 by matthijskooijman