arduino/Arduino #5473: IDE crash when a SAMD21 board is connected. JSON file or platform.txt parse bug?

RabidPrototypes, 2016-10-12:
I've designed a custom Zero compatible board with a ATSAMD21E17A and created a board definition to go along with it.  This definition doesn't differ much from the Arduino Zero's, yet half my users report that after installing the board files the IDE (1.6.8, 1.6.12, or any version in between) will crash if they load it while the board is connected, or if they connect the board after the IDE has been loaded.  

The only thing these users have in common is they are using Windows.  Some Windows 7, some Windows 10. I am using 10, and I do not experience the crash, so I have no way to diagnose the issue. 

I suspect there may be some sort of JSON parsing bug at play. Perhaps a buffer overrun. I think this because I don't know what other files in the board definition the IDE would be accessing when one simply connects the PCB. 

The JSON link to install my board files is here:
http://rabidprototypes.com/arduino/package_rabidprototypes_index.json

If I knew what files the IDE was accessing when a board is plugged in, I might have a better chance of diagnosing the issue. but aside from the JSON file, the only files I might expect it to access would be platform.txt.

RabidPrototypes, 2016-10-12:
One thing I have noticed and which occurred to me just now is that "Rabid Prototypes SAMD Boards (32bit ARM Cortex-M0+)" gets cut off on the boards menu.  Is it possible someone used a fixed-length array to store the board names, and this name being too long is causing a buffer overrun?

The only other thing I suspect at the moment is my use of "Tau / Firecricket" in the JSON file, but as far as I can tell a forward slash is allowed.

RabidPrototypes, 2016-10-15:
It appears this crash may actually occur regardless of whether the JSON files for my project are installed.  In other words, simply plugging in an ATSAMD21E17A board causes the latest IDE to crash.

PaulStoffregen, 2016-10-15:
Are you aware of arduino_debug.exe?  It lets you see the Java stack trace when the IDE crashes.  On Linux you get this automatically if running the IDE from a terminal.  But on Windows, you have to run the debug version.

RabidPrototypes, 2016-10-15:
Thanks for reminding me! 

While I can't replicate the problem myself and record a log of it, a customer sent me this log the other day.  I can't make heads or tails of it as I don't know Java, but maybe it will be obvious to someone who does what the problem is:

> Every time I plug either of my my Taus into my PC Running Arduino IDE version 1.6.12 (and 1.6.11)  (Win 7 64bit) It crashes with:

A fatal error has been detected by the Java Runtime Environment:

EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x617493cd, pid=4576, tid=0x000015b4

JRE version: Java(TM) SE Runtime Environment (8.0_92-b14) (build 1.8.0_92-b14)
Java VM: Java HotSpot(TM) Client VM (25.92-b14 mixed mode windows-x86 )
Problematic frame:
C  [listSerialsj.dll+0x93cd]

Failed to write core dump. Minidumps are not enabled by default on client versions of Windows

If you would like to submit a bug report, please visit:
http://bugreport.java.com/bugreport/crash.jsp
The crash happened outside the Java Virtual Machine in native code.
See problematic frame for where to report the bug.

---------------  T H R E A D  ---------------

Current thread (0x06054800):  JavaThread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" [_thread_in_native, id=5556, stack(0x07720000,0x07770000)]

siginfo: ExceptionCode=0xc0000005, writing address 0x00000000

Registers:
EAX=0x00000000, EBX=0x0000072c, ECX=0x00000000, EDX=0x00000001
ESP=0x0776eed0, EBP=0x0776f328, ESI=0x00000002, EDI=0x00611908
EIP=0x617493cd, EFLAGS=0x00010246

Top of Stack: (sp=0x0776eed0)
0x0776eed0:   007a26c8 0000002e 00000002 00000000
0x0776eee0:   00000003 00000000 00000000 02c8a31d
0x0776eef0:   128d0590 00000009 0000006c 763b27bd
0x0776ef00:   05bd0960 00000009 00611820 00003c18
0x0776ef10:   128d05a0 00000000 000018a8 00000200
0x0776ef20:   00000001 0000006c 00001438 00000001
0x0776ef30:   00000000 128d05a0 314d4f43 00ef0032
0x0776ef40:   00425355 04006d65 00000000 04ef01d8 

Instructions: (pc=0x617493cd)
0x617493ad:   8b 45 08 8b 90 94 00 00 00 85 d2 74 16 8b 40 2c
0x617493bd:   c7 44 24 04 2e 00 00 00 89 04 24 e8 73 14 00 00
0x617493cd:   c6 00 00 8d 85 04 fc ff ff 89 44 24 18 8d 85 0a
0x617493dd:   fc ff ff 89 44 24 10 a1 e4 01 75 61 c7 85 04 fc 

Register to memory mapping:

EAX=0x00000000 is an unknown value
EBX=0x0000072c is an unknown value
ECX=0x00000000 is an unknown value
EDX=0x00000001 is an unknown value
ESP=0x0776eed0 is pointing into the stack for thread: 0x06054800
EBP=0x0776f328 is pointing into the stack for thread: 0x06054800
ESI=0x00000002 is an unknown value
EDI=0x00611908 is an unknown value

Stack: [0x07720000,0x07770000],  sp=0x0776eed0,  free space=315k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [listSerialsj.dll+0x93cd]
C  [listSerialsj.dll+0x2211]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  processing.app.Platform.listSerialsNative()[Ljava/lang/String;+0
j  processing.app.Platform.listSerials()Ljava/util/List;+5
j  cc.arduino.packages.discoverers.serial.SerialBoardsLister.retriggerDiscovery(Z)V+22
j  cc.arduino.packages.discoverers.serial.SerialBoardsLister.run()V+9
[... 217 lines cut ...]
VM Arguments:
jvm_args: -Dsun.java2d.d3d=false -Djna.nosys=true -DAPP_DIR=C:\Program Files (x86)\Arduino -Xms128M -Xmx512M 
java_command: processing.app.Base
java_class_path (initial): C:\Program Files (x86)\Arduino\lib\apple.jar;C:\Program Files (x86)\Arduino\lib\arduino-core.jar;C:\Program Files (x86)\Arduino\lib\batik-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-anim-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-awt-util-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-bridge-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-codec-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-css-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-dom-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-ext-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-gvt-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-parser-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-rasterizer-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-script-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-squiggle-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-svg-dom-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-svgpp-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-transcoder-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-util-1.8.jar;C:\Program Files (x86)\Arduino\lib\batik-xml-1.8.jar;C:\Program Files (x86)\Arduino\lib\bcpg-jdk15on-152.jar;C:\Program Files (x86)\Arduino\lib\bcprov-jdk15on-152.jar;C:\Program Files (x86)\Arduino\lib\commons-codec-1.7.jar;C:\Program Files (x86)\Arduino\lib\commons-compress-1.8.jar;C:\Program Files (x86)\Arduino\lib\commons-exec-1.1.jar;C:\Program Files (x86)\Arduino\lib\commons-httpclient-3.1.jar;C:\Program Files (x86)\Arduino\lib\commons-lang3-3.3.2.jar;C:\Program Files (x86)\Arduino\lib\commons-logging-1.0.4.jar;C:\Program Files (x86)\Arduino\lib\commons-net-3.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-annotations-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-core-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-databind-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\jackson-module-mrbean-2.6.3.jar;C:\Program Files (x86)\Arduino\lib\java-semver-0.8.0.jar;C:\Program Files (x86)\Arduino\lib\jmdns-3.4.1.j
Launcher Type: SUN_STANDARD

Environment Variables:
CLASSPATH=.;C:\Program Files (x86)\Java\jre7\lib\ext\QTJava.zip
[... cut ...]
OS: Windows 7 , 64 bit Build 7601 (6.1.7601.23539)

CPU:total 2 (2 cores per cpu, 1 threads per core) family 6 model 23 stepping 6, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, sse4.1, tsc

Memory: 4k page, physical 8387960k(4805004k free), swap 16774060k(12698840k free)

vm_info: Java HotSpot(TM) Client VM (25.92-b14) for windows-x86 JRE (1.8.0_92-b14), built on Mar 31 2016 21:09:21 by "java_re" with MS VC++ 10.0 (VS2010)

time: Mon Oct 10 13:34:47 2016
elapsed time: 14 seconds (0d 0h 0m 14s)`

RabidPrototypes, 2016-10-15:
Of course, even not knowing Java, this line stands out to me like a sore thumb:
Current thread (0x06054800): JavaThread "cc.arduino.packages.discoverers.serial.SerialBoardsLister"

My first thought was that perhaps the name of my board in platforms.txt was too long and someone had shoved the names into a fixed length string. I noticed the name is too large to fit in the menu, but I hadn't had the chance to correct that and have someone test it when someone else reported that the crash happens regardless of whether the board files are installed.

It may be worth noting that I have the USB IDs set the same as those on the Zero (I chose to do this because I don't know how using a different one might affect the detection of the board by IDE with the included drivers on different versions of Windows), but I'm having a difficult time imagining why that would cause a crash, and I don't have any easy way of testing a different USB ID since I can't reproduce the crash myself.

PaulStoffregen, 2016-10-15:
Looks like a bug in listSerialPortsC.  Probably should open an issue here:

https://github.com/arduino/listSerialPortsC

While unrelated, you might also want to know about the CDC serial driver bug which exists in Windows 7 and all other versions before Windows 10.

https://www.youtube.com/watch?v=DRmvUsa2xuU

If another program has the COM port open during the disconnect event, when you later reconnect (as happens when reprogramming on these native USB ports) the COM port appears in the device manager, but can't be used.  It's a driver bug.  The only solution is to carefully avoid having the COM port open when uploading, which Arduino IDE does automatically, but other programs can't.  Or upgrade to Windows 10, or Mac or Linux.  All copies of Windows 7 have this annoying bug.  Still, the IDE (and listSerialPortsC) shouldn't crash.

RabidPrototypes, 2016-10-15:
That customer was using Windows 7, but the crash bug occurs in Windows 10 as well, so this issue is unrelated to the CDC driver bug.

facchinm, 2016-10-20:
Indeed, this problem is libSerialPort related. I'll try to reproduce it by flashing your board's bootloader into a Zero and see what happens on Win. Maybe the long name can be an hint, I'll investigate on that

facchinm, 2016-10-24:
@RabidPrototypes , I reproduced it on Win10 and fixed with https://github.com/arduino/Arduino/commit/de9bd89a01323d7358bfc1c20834e77a7642d3f4 . Could you please test latest nightly to confirm that it works now? Thanks

RabidPrototypes, 2016-10-27:
So far it seems to have fixed it! Thanks!

closed 2016-10-28 by facchinm