arduino/Arduino #5475: The program showing numbers in some code

daleyev, 2016-10-12:
https://i.gyazo.com/c781374b63e2963236f176f51bc6e934.png

Like showed in the photo the numbers i wrote are 1 to 0.
I've tried re-installing the program but still it didn't work for me for some reason...same with copying codes from notepad to the program.
I am using windows 10 /64 bit...

dlabun, 2016-10-13:
Are you using a standard keyboard in an English copy of Windows 10?

daleyev, 2016-10-15:
Yes...

facchinm, 2016-10-20:
It looks like a font problem, maybe a malicious replacement of a system font (`Monospaced` if `preferences.txt` is untouched). Try changing the font as explained here http://tminusarduino.blogspot.it/2016/03/how-to-theme-and-change-font-of-arduino.html and see if the problem presists

sandeepmistry, 2017-02-16:
Closing this for now due to lack of feedback.

closed 2017-02-16 by sandeepmistry