arduino/Arduino #5478: include overwrites symbolic links with copies of files

Copmic, 2016-10-13:
setup is:
/project1/project1.ino
/project1/localinclude1.h
/project1/include2.h
/include2.h

where /project1/include2.h is a symbolic link to /include2 and
/project1/localinclude1.h is a "real" file.

project1.ino includes both .h via 
#include "localinclude1.h"
#include "include2.h"

system is Linux Debian Jessie 64bit; arduino ide 1.6.11; file system is ext4

############

If I open /project1/project1.ino in arduino ide then all three files (project1.ino, localinclude1.h, include2.h) are shown in ide in tabs (= correctly).
I can edit and save all three files independently. If I save project1.ino or localinclude1.h from the ide all is fine and /project1/include2.h is still a symbolic link.

But if I save include2.h from the ide the symbolic link is replaces by a copy of /include2.h.
Thus, I then still have 
/project1/project1.ino
/project1/localinclude1.h
/project1/include2.h
/include2.h

but now /project1/include2.h is a "real" file (not a link any more) and is a copy of /include2.h.

This bug is really hard to recognize, as at first everything seems to be ok and continues to work.
Only if later you change something in /include2.h it won't show up in /project1/include2.h an v.v.

facchinm, 2016-10-20:
Hi @Copmic , I agree that soft links should be saved by following the link and saving the actual file. However, this scenario is quite uncommon and platform-dependant, so the solution may be less than obvious. I changed the title into something more explicit

closed 2016-12-07 by facchinm

Madouc, 2017-02-08:
this issue is also occurring within Windows 10 in Arduino 1.6.11.
Created files (both hard and sym link)
`mklink global.h ..\libraries\mcp2515\global.h`
or 
`mklink /h global.h ..\libraries\mcp2515\global.h`

Madouc, 2017-02-08:
Arduino 1.8.1 on windows 10 still exhibits this behavior.

symlinks created:
'mklink global.h ..\libraries\mcp2515\global.h

but overwritten on save.

45gfg9, 2019-07-29:
Problem consists to this day.
See #8535

PR #5573 (2016-12-07): Fix symlinks being replaced with files on save
Fixes #5478

tested on Linux, please confirm that it works in Win and OSX before merging
commit 2d1f49a0 (2016-12-07): Fix symlinks being replaced with files on save

Fixes #5478

Backported from Processing https://github.com/processing/processing/blob/0abee5af6ad3b11cf2b73bb794b8a97c157c4762/app/src/processing/app/Util.java#L174