arduino/Arduino #5487: Compiler error - java.lang.NullPointerException (arduino 1.6.12)

knefel, 2016-10-15:
I'm using the arduino compiler with the nxcEditor since nearly 1 year. Everything was good until I changed from arduino-1.6.6 to arduino-1.6.12. The following line is no problem with arduino-1.6.6, but with the new version there is a problem.

Tested inside a terminal window (by the way it's the example program Blink.ino):

/opt/arduino-1.6.12/arduino --board arduino:avr:uno --port /dev/ttyACM0 --upload Blink.ino

This is the command line output:

Picked up JAVA_TOOL_OPTIONS: 
Konfiguration wird geladen...
java.lang.NullPointerException
	at processing.app.BaseNoGui.selectSerialPort(BaseNoGui.java:1088)
	at processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)
	at processing.app.Base.<init>(Base.java:273)
	at processing.app.Base.guardedMain(Base.java:219)
	at processing.app.Base.main(Base.java:132)

I've tested with Ubuntu 12.04, Ubuntu 16.04 and Opensuse Leap 42.1 (32 bit and 64 bit).

per1234, 2016-10-15:
Duplicate of #5413, which has already been fixed in the hourly build. Please always test the hourly build before reporting an issue to check if the problem has already been solved and if not then search for issues/PRs to make sure the issue hasn't already been reported.

facchinm, 2016-10-20:
Thanks @per1234, closing as solved. Stable version containing this patch is on its way.

closed 2016-10-20 by facchinm