arduino/Arduino #5492: Compiler won't notice if files are deleted

rkost, 2016-10-15:
For version 1.6.11 the compiler throws a "multiple definition of `DummyClass::getString()'" error if one deletes the DummyClass files (DummyClass.h/.cpp) and copies it to e.g. the main .ino file.

Using the option "use external editor"

facchinm, 2016-10-20:
Maybe related to Editor refactor and files not getting actually deleted? @matthijskooijman

per1234, 2017-07-09:
@rkost, does this problem still occur for you with Arduino IDE 1.8.3?

rkost, 2017-07-14:
Ooops, sorry. I will test this as soon as I can. Didn't noticed your comment.

per1234, 2018-01-16:
@rkost, did you ever get a chance to check if this still occurs in the latest IDE version?

rkost, 2018-01-20:
So I just re-tested it, and yes: The compiler still does not care if I delete a file called e.g. "Log.h" and use macros (that are only defined in this very file) elsewhere. I just took a look into the temp directory that is obiously being used by this IDE (in my case it was `/tmp/arduino_build_761725/sketch`) and there are copies of all the files I deleted in the directory I was working in (e.g. `/home/rkost/repos/test`).

So it seems that the files are not watched properly. I did not do things like that before, but I think that one might just delete the file(s) in the temp directory when inotify sends the `delete_self` event for any of the sketch's files.

I don't know if this is important for you, but here are my system specs (hardware should be irrelevant?): Arch Linux 64-Bit with Kernel **v4.14.13-1** on Intel, Arduino Version 1.8.5. 

I also tried this in one of my virtual machines, just to be sure that there is no special shit going on my personal machine. Was able to reproduce it on Ubuntu 17.10, so it seems to be a common problem on Linux. I will have a look at windows 10 on Sunday.

rkost, 2018-01-20:
Uh. And one more thing: The actual Editor gui **is updating properly**. If I delete a file `Log.h` in the working dir, it won't be shown in the file tabs at the top of the editor. It also properly disappears if I restore that file.

facchinm, 2018-01-22:
Hi @rkost, I'd ask you to test the [beta version](https://www.arduino.cc/en/Main/Software#beta) of the IDE (1.9) and report if this behaviour is solved. We added both a watcher on the sketch folder which should work in both "external editor" and normal mode.

matthijskooijman, 2018-01-23:
This is a problem in the builder: It copies all files into the temp directory, but does not clean out the temp directory beforehand. I've been planning to fix this by skipping the copy altogether and compiling straight out of the original sketch directory, but haven't gotten around to implementing that yet...

rkost, 2018-01-23:
Hey @facchinm,
I tested it on the beta version (1.9-build-35) and as @matthijskooijman said, the bug still persists. The file won't be deleted in the temp directory if I remove it inside the normal project directory.

Thanks by the way for all the time you are spending on this project!

facchinm, 2018-01-24:
MAYBE I got it. Is your sketch (or any of its parent folders) a symlink? I just noticed that the Beta watcher doesn't follow symlinks so your bugreport would still apply.

rkost, 2018-01-24:
Hmm, no. My project does not contain any symlinks. Also the path to the project does not consist of any symlinks (It is as simple as `/home/rkost/repos/company/projectName`). I don't know if the watcher has problems watching project folders that conain subdirectories. In my case:

```
├── config
│   └── robot.dimension.h
├── constants.h
├── JointCalibration.cpp
├── JointCalibration.h
├── Joint.cpp
├── Joint.h
├── Leg12X.cpp
├── Leg12X.h
├── Leg.h
├── LICENSE
├── Log.h
├── Motor.cpp
├── Motor.h
├── README.md
├── src
│   ├── calibrationUI
│   │   ├── CalibrationViaConsole.cpp
│   │   └── CalibrationViaConsole.h
│   ├── Module.h
│   ├── ModuleHandler.cpp
│   ├── ModuleHandler.h
│   ├── serialRemoteControl
│   │   ├── SerialRemoteControl.cpp
│   │   ├── SerialRemoteControl.h
│   │   ├── SerialRemoteInterface.cpp
│   │   └── SerialRemoteInterface.h
│   ├── servo
│   │   ├── Servo.cpp
│   │   ├── Servo.h
│   │   └── ServoTimers.h
│   └── walkingEngines
│       ├── WalkingEngine124.cpp
│       ├── WalkingEngine124.h
│       ├── WalkingEngineGeneric.cpp
│       └── WalkingEngineGeneric.h
├── tools
│   └── Z12x_3.ggb
├── UltrasonicSensor.cpp
├── UltrasonicSensor.h
└── Zuri.ino

```
How does the watcher handle unknown files? The sketch also contains a *.git* dir...

facchinm, 2018-01-24:
Ah, now I get the point. The watcher only watches the first folder level to update the tabs on demand. It doesn't watch any subfolder currently. I'll prepare a PR to address it

facchinm, 2018-01-25:
@rkost latest version of the beta build contains https://github.com/arduino/arduino-builder/pull/250/commits/849faa1f40336d51c9b16726213619e3d01cc19d which targets your issue (considering subfolders changes at build time). 
However, the problem is broader, so @matthijskooijman suggestion to use the actual source folder as backing storage would solve every dependency issue, while avoiding wiping the whole project folder every time something changes. I'll take a look at that approach too and report back.
If you could test the beta in the meantime and report if the issue is solved it would be great.

matthijskooijman, 2018-01-25:
@facchinm, I have already given some thought on the compile-from-sketch-folder approach, so perhaps ping me on slack if you want to discuss this?

rkost, 2018-01-25:
Nice!
Just tested it on the sketch mentioned above (see file tree). Compiled first, then deleted Log.h and let the ide build again. Compilation failed (as expected):

```
/home/rkost/software/arduino-PR-beta1.9-BUILD-36/arduino-builder -dump-prefs -logger=machine -hardware /home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware -tools /home/rkost/software/arduino-PR-beta1.9-BUILD-36/tools-builder -tools /home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/tools/avr -built-in-libraries /home/rkost/software/arduino-PR-beta1.9-BUILD-36/libraries -libraries /home/rkost/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10900 -build-path /tmp/arduino_build_762984 -warnings=all -build-cache /tmp/arduino_cache_517280 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/tools/avr -verbose /home/rkost/repos/x/Zuri/Zuri.ino
/home/rkost/software/arduino-PR-beta1.9-BUILD-36/arduino-builder -compile -logger=machine -hardware /home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware -tools /home/rkost/software/arduino-PR-beta1.9-BUILD-36/tools-builder -tools /home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/tools/avr -built-in-libraries /home/rkost/software/arduino-PR-beta1.9-BUILD-36/libraries -libraries /home/rkost/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10900 -build-path /tmp/arduino_build_762984 -warnings=all -build-cache /tmp/arduino_cache_517280 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/tools/avr -verbose /home/rkost/repos/x/Zuri/Zuri.ino
Using board 'mega' from platform in folder: /home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/arduino/avr
Build options changed, rebuilding all
Detecting libraries used...
"/home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/arduino/avr/cores/arduino" "-I/home/rkost/software/arduino-PR-beta1.9-BUILD-36/hardware/arduino/avr/variants/mega" "/tmp/arduino_build_762984/sketch/Zuri.ino.cpp" -o "/dev/null"
/home/rkost/repos/x/Zuri/Zuri.ino:17:17: fatal error: Log.h: No such file or directory
 #include "Log.h"
                 ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

Afterwards I restored the file, and the IDE compiled again as expected. Also tried this with `src/WalkingEngines/WalkingEngineGeneric.h` with the same results. Seems to work now.

So I have to thank you for that great support!