arduino/Arduino #5493: Compiler warnings not shown if compilation was done a second time

rkost, 2016-10-15:
When someone writes code like this (see below) he will correctly get two identical warnings: 
**```"warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]"```**

When one compiles it again the warnings will all **dissappear** so the user (me) thinks he is perfectly fine and writing clean code. 
**FunFact:** This will **not** happen if all code is written to the main (.ino) file.

File "DummyClass.h":
```C++
class DummyClass
{
public:
  char* getString();

  char* getSecondString();
};
```

File "DummyClass.cpp":
```C++
char* DummyClass::getString()
{
  return "Some String";
}

char* DummyClass::getSecondString()
{
  return "Some other string";
}
```

File "test.ino":
```C++
#include "DummyClass.h"
DummyClass dummy;

void setup()
{
  Serial.begin(115200);
}
void loop()
{
  Serial.println(dummy.getString());
  delay(1000);
}
```

This is why you want to accept #5361 or setup an option in preferences to activate compiler errors when warnings occur.

matthijskooijman, 2016-10-17:
This is caused by caching: the second time the file is not recompiled if it was unchanged, so the warnings are not shown. The only way I can see to fix this (other than making warnings into errors), is to cache the stdout/stderr output as well, so it can be shown again even when not running the compiler. Not sure if this is the best way to approach this, though.

facchinm, 2016-10-20:
@matthijskooijman  is correct, caching elides warnings from previous compilations. I believe that `ccache` caches stdout/stderr and replays them even if the cached version is unchanged, so the approach could be ok, we only need to find a suitable "backend" for storing this information (taking care it also works in a cloud infrastructure)