arduino/Arduino #5496: New IDE update notice missing?

NicoHood, 2016-10-16:
If a new library is available the IDE will prompt you for an update. But this does not seem to happen for the IDE itself, but I remember that this worked before.

Is it just because I recompiled the IDE myself, or does this only happen on linux?

And if it works again, how can I deactivate it? I've found this, but is this still the correct way to do so?
```
    # echo -e "\n# By default, don't notify the user of a new upstream version." \
    #         "\n# https://bugzilla.redhat.com/show_bug.cgi?id=773519" \
    #         "\nupdate.check=false" \
    #     >> build/shared/lib/preferences.txt
```

per1234, 2016-12-26:
I believe the cause of the updatable IDE notification no longer being shown is that https://www.arduino.cc/latest.txt has not been updated since 1.6.1.

See: https://github.com/arduino/Arduino/blob/master/app/src/processing/app/UpdateCheck.java#L54

oqibidipo, 2017-02-21:
At last! `latest.txt` says 10801.

facchinm, 2017-02-21:
Yeah, we updated it last week :smile: Forgot to close this issue, thanks for the remainder

NicoHood, 2017-02-21:
I do not get an update notification for archlinux. This is okay, as we dont want a notifications, however I am not sure if the feature is broken then?

oqibidipo, 2017-02-26:
The problem is that the update URL in the IDE is an HTTP URL which is redirected to HTTPS. HttpURLConnection will not follow the redirect when the protocol changes. Changing the URL in the source will fix this for new new versions, but old versions will still remain silent about updates.

However, this could be solved by configuring the server to deliver `latest.txt` over HTTP while redirecting everything else.

NicoHood, 2017-02-27:
Nah I think the webserver should stay with https. However it is good to know why this happens.

facchinm, 2017-02-27:
@oqibidipo @NicoHood thanks for the heads-up and the debugging session, we'll check what we can do to get the notification back on older IDEs (I think serving that file over http is ok)

SasmitoMotion, 2017-02-27:
Thanks mr that IDE to border arduino mega and wifi

per1234, 2017-02-27:
This is the third OT/spam comment I've seen from @SasmitoMotion:
https://github.com/arduino/Arduino/issues/6020#issuecomment-282591639
https://github.com/arduino/Arduino/pull/5990#issuecomment-282736143
There are probably more. Someone please block them.

@SasmitoMotion please stop, you are spamming hundreds of people.  If you need help with something then post a question to the [Arduino Forum](http://forum.arduino.cc/).  Make the effort to explain yourself completely and clearly. There may be a forum section for your native language.

per1234, 2017-05-10:
It looks like https://www.arduino.cc/latest.txt has reverted to 10601 for some reason.

facchinm, 2017-05-10:
@mastrolinux could you check please? Thanks @per1234

per1234, 2017-06-16:
@mastrolinux https://www.arduino.cc/latest.txt is still at 10601. I just don't understand how this can be so difficult.

mastrolinux, 2018-10-15:
I do not know what string I write in the file, please advice @facchinm

facchinm, 2018-10-15:
@mastrolinux `10807` will be fine ;)

per1234, 2019-02-07:
I'm reopening because https://www.arduino.cc/latest.txt was never updated to 10808 when the release of Arduino IDE 1.8.8 happened.

Can this be automated or added to a official release checklist so that it doesn't get forgotten on future releases?

facchinm, 2019-02-11:
@mastrolinux may you update that file to  read `10808` ?

alranel, 2019-05-08:
I'm running 1.8.8 (macOS) and I'm not getting the update notice.
http://www.arduino.cc/latest.txt currently contains `10809` and updates are enabled in Preferences. Not sure what's going on here...

per1234, 2019-05-10:
>I'm running 1.8.8 (macOS) and I'm not getting the update notice.

I also don't get update notices with Arduino IDE 1.8.8 running on a Linux 64 bit virtual machine or the Linux ARM 32 bit version on a Raspberry Pi.

I do get the expected update notice with Arduino IDE 1.8.8 on Windows 10.

ngammarano, 2020-02-20:
https://www.arduino.cc/latest.txt currently says `10810`.

Shouldn't it say `10812` as it is the latest version (https://www.arduino.cc/en/Main/Software)?

oqibidipo, 2020-08-30:
1.8.13 has been out for two months, but latest.txt says `10812`.

rsora, 2020-08-31:
@oqibidipo https://www.arduino.cc/latest.txt should show `10813` now.

Happy coding!