arduino/Arduino #5497: linino yun ethernet link down

esaid, 2016-10-16:
Hello,

On my linino yun , why the Ethernet link down after few minutes .

when the linino is ok :

root@linino:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         r8-r-rhpzm-1-pu 0.0.0.0         UG    10     0        0 eth1
128.141.161.0   *               255.255.255.0   U     10     0        0 eth1
192.168.240.0   *               255.255.255.0   U     0      0        0 wlan0
 

and after 20-30 min , under terminal i can see :

root@arduino:/# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
128.141.161.0   *               255.255.255.0   U     10     0        0 eth1
root@arduino:/#
root@arduino:/#

I can restart the network (under terminal ) :

/etc/init.d/network restart
[ 3328.510000] eth1: link down
[ 3330.370000] eth1: link up (100Mbps/Full duplex)
 

this is my linino yun :
root@linino:~# /usr/bin/lininoio -v
DISTRIB_ID="Linino"
DISTRIB_REVISION="223d5a8"
BUILD_DATE="2016-06-01"
 
 
can you help me , anybody have the same problem ?

On my arduino yun , i dont have this problem , but just on linino board.

Cheers

Emmanuel

facchinm, 2016-10-20:
Hi Emmanuel,
the right repo to report this is Linino's one (https://github.com/linino). Please file an issue there because the solution could be non trivial (or maybe it's already known)

closed 2016-10-20 by facchinm

esaid, 2016-10-20:
ok,

so the solution is  add in  /lib/netifd/dhcp.script the line   **route add -net 0.0.0.0/0 gw "$i"**

cheers

Emmanuel

setup_interface () {
        proto_init_update "*" 1
        proto_add_ipv4_address "$ip" "${subnet:-255.255.255.0}"
        # TODO: apply $broadcast

        for i in $router; do
                proto_add_ipv4_route 0.0.0.0 0 "$i"
--->>       route add -net 0.0.0.0/0 gw "$i"
        done