arduino/Arduino #5499: Uploading: The system cannot find the file specified

StormTrooper, 2016-10-16:
I've got a arduino mini with a FTDI usb converter. 
On IDE 1.6.9 it works fine. I've just downloaded and installed 1.6.12 and am getting:

Quote
java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
 at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
 at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
 at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
 at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:206)
 at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
 at processing.app.SketchController.upload(SketchController.java:707)
 at processing.app.SketchController.exportApplet(SketchController.java:680)
 at processing.app.Editor$DefaultExportHandler.run(Editor.java:2176)
 at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
 at java.lang.ProcessImpl.create(Native Method)
 at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
 at java.lang.ProcessImpl.start(ProcessImpl.java:137)
 at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
 ... 8 more
An error occurred while uploading the sketch

Going back to 1.6.9 and it works again.

Another user having the same issue:

https://forum.arduino.cc/index.php?topic=428167.msg2951183#msg2951183

Windows 10
64 Bit

facchinm, 2016-10-20:
Hi @StormTrooper ,
this looks like a problem with older cores overriding newer ones. Please post the AVR core version shown on "Board Manager" window and the full compilation verbose output on 1.6.12. Eventually, you can force using the bundled AVR core by removing `C:/Users/YourUser/AppData/Local/Arduino15/packages/arduino/hardware/avr/` folder

per1234, 2017-08-22:
@StormTrooper we're still waiting on you to provide the information facchinm requested.

per1234, 2018-01-16:
Closing due to lack of response from @StormTrooper.

closed 2018-01-16 by per1234