arduino/Arduino #5502: Selecting "Windows 10 Iot Core" gives null pointer exception

bfjelds, 2016-10-17:
Seems that the Win10 boards.txt lacks a "buld.core" property and this code throws a null pointer exception:

Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
              at processing.app.BaseNoGui.getBoardPreferences(BaseNoGui.java:188)

Seems to be in this code ... assume Win10 plugin needs to be fixed, perhaps this should as well:

    // Add all tools dependencies from the (possibily) referenced core
    String core = prefs.get("build.core");
    if (core.contains(":")) {

facchinm, 2016-10-20:
@cmaglie need to add a check for NPE (and inform who is in charge for Win10 IoT board to fix the problem).
@bfjelds , thanks for reporting!

bfjelds, 2016-10-20:
No problem!  I'm a part of the Win10 IoT team, we have a new version 1.1.1 that contains a fix.

commit 12eb09f0 (2016-11-09): Fix NPE if referred core is nonexistent

Fixes #5502
closed 2016-11-09 by cmaglie