arduino/Arduino #5503: [Library Manager] Checksum (CRC) doesn't match. File corrupted.

HackerInside0, 2016-10-17:
Last friday i have noticed that if i try to download my own libraries from the library manager, the download fails and the error that i have reported in the title appears.

Before open this issue, i've waited for the library manager refresh on sunday, but it doesn't had any effect.

I'm talking about the L293 library:
https://github.com/HackerInside0/Arduino_L293.git

and the SharpIR library:
https://github.com/Hackerinside0/Arduino_SharpIR.git

[image]

cmaglie, 2016-10-18:
It's working for me...

[image]

maybe you already downloaded some files that do not match crc?
You can try one of the following:

- remove the folder `$HOME/.arduino15/staging/libraries/`

or

- try the hourly build of the Arduino IDE that should do this automatically (this is a recent fix that will be released with IDE 1.6.13)

HackerInside0, 2016-10-18:
Resolved, thanks.

closed 2016-10-18 by HackerInside0