arduino/Arduino #5504: Outdated bootloader hex files

lfdebrux, 2016-10-18:
Although the source code for `atmega` and `bt` bootloaders was updated a year ago, the compiled hex files date back 5 years.

[image]

This turned out to be a problem for me because the Arduino Pro Mini bootloader that is pre-built seems to differ in a way from the latest source code in a way that causes issues with uploading code over serial.

Is there any reason not to create a pull request with newly compiled files that are up to date with the source code?

facchinm, 2016-10-20:
No particular reason I believe except the fact that ATmegaBOOT has been replaced by `optiboot` a long time ago (see here for a comparison http://forum.arduino.cc/index.php?topic=159247.0). 
The precompiled `hex`s in there are the production binaries so they stopped being updated when the boards started being factory flashed with optiboot.
The correct bootloaders you are looking for are probably here https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/bootloaders/optiboot
I'm closing this issue, reopen if needed

closed 2016-10-20 by facchinm

ortegafernando, 2018-09-17:
[](url)