arduino/Arduino #5505: File, Examples, 'order' issue

4dsystems, 2016-10-19:
We have become aware of an 'order' issue with the way examples are shown in the IDE when examples that are both applicable to 'any board' and are also included in a JSON for a specific board.

Rather than the libraries showing as being applicable for 'any' board, they are only shown for the specific board.

Prior to the JSON being installed, the library examples would be listed under 'any board' as should be the case, but after the JSON is installed, that common library moves to be only under the specific board that was installed with the JSON. 

We also noticed that when the project is compiled, the compiler detects a duplicate library, and decides which one it is going to use. 

For example, our JSON file includes the EEPROM library which is included in the standard Arduino release for the Leonardo (for example). When we add the JSON file, the EEPROM library gets removed from the listing from 'All Boards' and is then only listed under our board installed by the JSON.

This should not be the case, as EEPROM is used by multiple boards, not just our JSON, yet the listing indicates its only applicable to our one specific JSON installed product.

cmaglie, 2016-10-19:
> For example, our JSON file includes the EEPROM library which is included in the standard Arduino release for the Leonardo (for example). When we add the JSON file, the EEPROM library gets removed from the listing from 'All Boards' and is then only listed under our board installed by the JSON.

I'm not sure that I understand your issue correctly, but the following is a screenshot from an installation without third party json for AVR boards:

[image]

as you can see EEPROM is listed under libraries for Leonardo board. Moreover EEPROM is not a library for any board, becuase it's an architecture specific library and it can be used only on AVR based board (so it won't work on Arduino Zero / Due / 101 for example).

4dsystems, 2016-10-20:
EEPROM was just an example and maybe not a good one. Say the Ethernet Library, which is currently listed under 'Examples for Any Board'. If you were to have this same library installed with a JSON file, the Ethernet examples would no longer be listed under 'Examples for Any Board', and it would only be listed under the product with the JSON you just installed. 

Here is a more specific example.
We have a serial library for our 4D display modules. We have adapted our own IDE to utilise the arduino_debug.exe file to compile examples using the Arduino IDE, but inside our own IDE. When we install our IDE, it automatically installs a few 'standard' libraries of ours, as they are used for numerous boards, and they get installed by our software into the "C:\Program Files (x86)\Arduino\libraries" area on a PC, which is like a built in library like Ethernet I guess. This is done so they are in a consistent location, as if we put it in AppData and Arduino15 folders, then one day this may change and break everything, and I am not even sure it works from there as that is for the Board Manager installed things...  
If we then have a JSON file installed for a specific board which happens to use the same library as the standard one we installed with our IDE, the original library located in program files gets removed from the 'Examples for any board' area, and it is only specifically listed under the JSON board.

We get into this position as our IDE only works for Windows, So if a customer uses a MAC for example, then the library needs to be installed with the JSON too.

Now you are probably going to say, don't install the libraries into Program Files. This isn't really a solution though, as this simply shouldn't happen. The JSON file should not cause example libraries to disappear from the 'Examples for Any Board' and only show up under this one specific JSON installed board, as its relevant for both, so it should stay installed for the 'Examples for Any Board'.

Does that make sense?

per1234, 2017-07-09:
The **File > Examples** menu should reflect the library override system and as far as I'm aware it does.

Libraries bundled with a hardware package are only accessible when one of the boards of that package is selected and the library bundled with the hardware package will have include priority over the one bundled with the Arduino IDE. Therefore when that library has include priority it is the correct behavior for the examples to appear under **File > Examples > Examples for ...**. 

However, libraries installed to the sketchbook folder have include priority over libraries bundled with hardware packages. Since https://github.com/arduino/arduino-builder/pull/123 this is only done if the architecture value in their library.properties file is an equal or better match to the architecture of the selected board. So if you have in the library.properties file for the library bundled with your hardware package:
```
architectures=*
```
and in the library.properties file of your library installed to the {sketchbook}/libraries folder:
```
architectures=*
```
the examples for the library will show up in **File > Examples > Examples from Custom Libraries**.

>Now you are probably going to say, don't install the libraries into Program Files.

OK, I'll say it. Don't do that. If possible don't mess with the Arduino IDE installation at all. Arduino has made it very easy to add support for 3rd party hardware to the Arduino IDE using Boards Manager. It's always a nightmare trying to help people on the forum when they bought some hardware that required modifications to the Arduino IDE installation and every time I've encountered that it was completely unnecessary. Install libraries to the {sketchbook}/libraries folder. You can even make them super easy for the users to install to that location by adding them to the Library Manager index:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

closed 2017-07-09 by per1234