arduino/Arduino #5511: Arduino IDE crashes when Arduino is plugged in.

minswho, 2016-10-20:
IDE crashes immediately when I plug in my Arduino Micro.
IDE crashes in the middle of boot screen when Arduino is already plugged in and leages a "hs_err_pidXXXX" log in the projects folder.

> A fatal error has been detected by the Java Runtime Environment:
EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x617493cd, pid=4080, tid=0x000008f8
JRE version: Java(TM) SE Runtime Environment (8.0_92-b14) (build 1.8.0_92-b14)
Java VM: Java HotSpot(TM) Client VM (25.92-b14 mixed mode windows-x86 )
Problematic frame:
C  [listSerialsj.dll+0x93cd]
Failed to write core dump. Minidumps are not enabled by default on client versions of Windows

I tried everything from power cycling to reinstalling the IDE itself.

I'm honestly baffled at this problem and would appreciate any help.

facchinm, 2016-10-24:
Would you mind testing the latest nightly? It contains https://github.com/arduino/Arduino/commit/de9bd89a01323d7358bfc1c20834e77a7642d3f4 which should fix the issue

facchinm, 2016-10-24:
Hi @minswho ,
could you test latest hourly build of the IDE and confirm that it's working now?

minswho, 2016-10-24:
Hourly build seems to have solved the issue.
Thanks facchinm!

closed 2016-10-25 by facchinm