arduino/Arduino #5513: Add RGBLEDBlender library

ESikich, 2016-10-22:
Hi, I've created a library for easy color blending on an RGB LED.

https://github.com/ESikich/RGBLEDBlender.git

Thanks much,
Erik

agdl, 2016-10-26:
Hi please fix library.properties according to the tag

https://github.com/ESikich/RGBLEDBlender/blob/1.1.3/library.properties

ESikich, 2016-10-26:
Fixed.

cmaglie, 2016-11-02:
I've added the library to the library manager, btw the problem with release 1.1.3 is not fixed.
The library.properties and the tag do not match, as you can see here:

https://github.com/ESikich/RGBLEDBlender/blob/1.1.3/library.properties

the tag is "1.1.3" but library.properties shows "1.1.2".

You can fix this by strictly following these steps:

1) remove the tag 1.1.3
2) fix library.properties and set version to 1.1.3
3) commit and push
4) re-add the tag 1.1.3 (**after** doing 2 and 3)

Once done, the change is automatically picked up by our servers within an hour, no need to further comment here.

closed 2016-11-02 by cmaglie