arduino/Arduino #5516: [Add library] CD74HC4067

waspinator, 2016-10-23:
https://github.com/waspinator/CD74HC4067

agdl, 2016-10-26:
Done thank you!

closed 2016-10-26 by agdl

waspinator, 2016-11-07:
@agdl I can't seem to find my library in the `Library Manager`. Maybe the way I prepared my library was incorrect? It's my first library I tried to add, and I'm not sure if I did everything right. Thanks.

cmaglie, 2016-11-09:
The crawler log reports:

`2016/11/09 13:37:25 'sentence' and 'url' fields must be defined`

Please update your library.properties by adding the missing fields, commit and push the change, and redo the 1.0.0 tag.
Once done, if everything is ok, the library should be automatically published within an hour.

waspinator, 2016-11-26:
@cmaglie I tried to update it as you described, but it's still not appearing in the Library Manager. Is there a place I can see what the log error reports?

cmaglie, 2016-11-26:
I can see it on the library manager:

[image]

It may be a connection or file permission problem?

waspinator, 2016-11-30:
@cmaglie I can see it now too. strange. thanks for looking into it. sorry for the trouble.