arduino/Arduino #5517: library manager not updating display to reflect properties for selected version

bperrybap, 2016-10-23:
When looking at a library in the library manager it appears that the library manager is not using the sentence and paragraph properties for the version of the library selected. It appears to be using the last one it sees for the library in the library_index.jason file rather than the property values of the version of the library selected.

I tested this using my hd44780 library.
I disabled network access to prevent a refresh of the jason file then I modified each sentence and paragraph in the jason file for the hd44780 library to indicated its version number.
When I brought up the library manager it showed the last one (0.7.2 when I did the testing)
And when selecting other versions of the library, the sentence or paragraph text shown did not change.

It seems like the expected behaviour when selecting a version of a library should be that the library manager updates the GUI display to reflect the property values being displayed for that version of the library rather than leave them based on a specific version of the library that is no longer selected. This would include updating author, name, sentence, paragraph, etc....

per1234, 2016-10-24:
Related Boards Manager issue: https://github.com/arduino/Arduino/issues/3329

bperrybap, 2017-01-08:
This is still an issue in 1.8.0 but there are some other unexpected subtleties.

You can now see this easily by using the hd44780 library.
The paragraph shown appears to be the one from the most recent version.
In version 0.8.3 I changed the paragraph.
If you select a version prior to 0.8.3 you will still see the 0.8.3  paragraph.

There is also some odd/unexpected search behavior.
If you search for something that is only in the most recent version, it finds the library ok, but it will not show any of other versions for selection.
So it appears that the search looks for the selected filter words and populates the version list based on the library versions that contain that search filter.

Example:
search for **bperrybap**
It finds the hd44780 library and shows the paragraph property from a version that contains it, but offers no ability to select any versions to install.
I'm guessing that is because at this point in time only 0.8.3 has **bperrybap** in it so there are no other versions that contain this filter to install.

search for **extensible**
It finds the hd44780 library and creates a dop down box with all the versions.
I'm guessing this is because all the versions contain the string **extensible**

search for **obsolete**
This string does not exist in the latest version of the properties (0.8.3 at this moment).
What is interesting is that that the paragraph property shown is not from the latest version but rather some other version that does have **obsolete** in it.

This is odd behavior, particularly considering that GUI display is not ever updated with the property values when a version of the library is selected from the drop down box.