arduino/Arduino #5518: Arduino IDE Accessibility problems

krperry, 2016-10-23:
First I want to say that the accessibility of Arduino IDE has got better over the last 12 releases.  With that said I am a blind Software Engineer who also develops hardware.  I am currently building a Bluetooth Probe.  While doing that I have found both of my screen readers NVDA and Jaws can not access the library manager, board manager, or the update window.    It is important that these things be accessible by the keyboard so that Blind and low vision as well as people with other disabilities that can not use the mouse can use these dialogs.    These are the only three dialogs I have found so far that II can not use in 1.6.12 but there may be others.  For any developer who wants to try out how this Arduino IDE works for a blind user they can download the "Non visual desktop access" software by searching for that text on google and downloading it.  It is free and is a great tool to make sure things are accessible.    If anyone knows an easy way for me to update boards with out the update manager please let me know.  kperry@blinksoft.com

LuisFerduino, 2018-06-05:
Hi, I'm an electronics engineer and I have profound visual weakness. I have already programmed microcontrollers by commands previously and the interfaces of the compilers worked correctly with the screen readers I have. (JAWS, NVDA AND NARRATOR.) But now when I entered Arduino, I found that none of them read me the Arduino IDE interface. So far I write all the code in the note pad and then I have to ask someone for support to help me compile it and upload it, making me dependent on other people to be able to do every test I need. I hope you can make the necessary improvements to make this platform accessible or recommend me some method to load my stages on the boards by my self and thus continue to developing applications in these microcontrollers.

krperry, 2018-06-05:
If you go to blindelectronics.com there is a blog on how to set it up.  You have to configure your access bridge correctly for it to work.  Note that someday I hope that Arduino IDE goes away from Swing and uses native controls so that screen readers can read it without the access bridge but as it is it is accessible with Jaws and NVDA as long as you setup the access bridge.  I am also 100% blind 

 

Ken

LuisFerduino, 2018-06-12:
hello, the procedure you gave me worked perfectly. I also found several interesting articles such as the talking multimeter that I did not know existed. thank you very much for your response and your willingness to help me solve this problem.

krperry, 2018-06-13:
No worries once I get time I am going to start blogging again.

Ken

tigoe, 2018-09-28:
I've been able to duplicate all of the issues reported by @krperry on MacOS. I haven't yet tried the access bridge solution he mentioned, but it's on my agenda to do. As a workaround, we've been using the arduino-cli, with a few extra tricks thrown in, [detailed at this link](https://itp.nyu.edu/physcomp/resources/using-the-arduino-command-line-interface/). I'd definitely advocate for a move away from swing on the next major upgrade.

Also, the pop up dialog that tells you when there are board or library updates does not work with the VoiceOver on MaOS controls. The close button will not close when you press the key combination to close it.