arduino/Arduino #5520: Problem using programmer "Arduino as ISP" on Linux ARM version

paologai, 2016-10-24:
If you try to program an Arduino with "Arduino as ISP" programmer you obtain the following error under Linux ARM version of the IDE:

**avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x15

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x01
avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x10
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): unknown response=0x12
La porta seriale selezionata avrdude: stk500_disable(): unknown response=0x12
 non esiste o la scheda non è connessa**

If you try to do the same thing (with the same source code)  under Linux (x86) or Windows O.S. it works great. It's only an ARM version bug.

isparkes, 2019-03-12:
Also on Linux Intel

paologai, 2019-03-12:
Grazie isparkes,
Come da te suggerito funziona alla grande!!!

closed 2019-03-12 by paologai