arduino/Arduino #5521: Documentation for file.openNextFile() issues

marcellosekoot, 2016-10-24:
http://assiss.github.io/arduino-zhcn/cn/Reference/FileOpenNextFile.html

This page has multiple issues:
1. The posted example code does not produce a useful result and does not illustrate how to properly use the file.openNextFile() method.
2. The Get Code link does not work (404).
3. The Edit Page | Page History | Printable View links do not work.

I would fix this myself if the Edit Page link worked.

per1234, 2016-10-25:
>http://assiss.github.io/arduino-zhcn/cn/Reference/FileOpenNextFile.html

That's not the official Arduino reference page, it's a scrape or something. Please refer to the official page:
https://www.arduino.cc/en/Reference/FileOpenNextFile

If you do want to address any issues that may exist on that unofficial page then you should open an issue report or pull request on the appropriate repository:
https://github.com/assiss/arduino-zhcn

>The posted example code does not produce a useful result and does not illustrate how to properly use the file.openNextFile() method.

The example code on the official page is different, please review the official example code and if there are any issues with it that is an appropriate thing to report here.

>The Get Code link does not work (404).

It works fine on the official page

>The Edit Page | Page History | Printable View links do not work.

The official page doesn't have those links.

facchinm, 2016-10-25:
Thanks @per1234 for pointing out the correct references. Closing as invalid

closed 2016-10-25 by facchinm

marcellosekoot, 2016-10-25:
My mistake, I’ll ignore that site. Thank you!

http://assiss.github.io/arduino-zhcn/cn/Reference/FileOpenNextFile.html

https://www.arduino.cc/en/Reference/FileOpenNextFile

https://github.com/assiss/arduino-zhcn