arduino/Arduino #5523: Intel Galileo Boards won't install on python3 linux distributions

NicoHood, 2016-10-25:
On Archlinux python3 is the default python. Possibly also on other distros sooner or later.

The file that gives the error is:
`.arduino15/packages/Intel/tools/i586-poky-linux-uclibc/1.6.2+1.0/i586/relocate_sdk.py`

The fix is simple:
```diff
-#!/usr/bin/env python
+#!/usr/bin/env python2
```

I could not find the galileo files on github, so I report that here. Where can i find the sources for this except for my local copy?

facchinm, 2016-10-26:
Hi Nico,
this is Galileo's core repo (https://github.com/01org/corelibs-galileo), unfortunately they closed the issue tracker so I'll share your report with them. Thanks !

facchinm, 2016-10-26:
@eriknyquist can you take a look at this (or escalate it if needed) ? Thanks

NicoHood, 2016-10-26:
The source for that python script is not in the repository. Thatswhy I though it is the wrong place. Otherwise I could fix it via PR. However I do not know if more problems will occur, but we will see.

eriknyquist, 2016-10-26:
@calvinatintel, where is that python script tracked?

NicoHood, 2016-12-13:
@eriknyquist @facchinm @calvinatintel any updates?

altf4arnold, 2018-02-17:
It seems that's not the only error. You might also have to change the install_script.sh at line 106.

Change the +111 to /111
it solved the problems for me