arduino/Arduino #5527: Library: Remove Adafruit_DHT_Unified

microbuilder, 2016-10-26:
Can you please remove the following library from the library manager: https://github.com/adafruit/Adafruit_DHT_Unified

We've merged both of our DHT libraries into a single repo to avoid confusion, thanks.

Timon2, 2016-10-27:
In file included from C:\Users\Laborator\Desktop\New_sketch_cu_do_while\New_sketch_cu_do_while.ino:2:0:

C:\Users\Laborator\Documents\Arduino\libraries\DHT_sensor_library/DHT_U.h:25:29: fatal error: Adafruit_Sensor.h: No such file or directory

 #include <Adafruit_Sensor.h>

                             ^

compilation terminated.

exit status 1

This error apears after library update

PLS HELP

microbuilder, 2016-10-27:
This is the wrong place to post support issues for Adafruit libraries, but you are missing the Adafruit_Sensor library.

Timon2, 2016-10-27:
thx, i instaled Adafruit_sensor-master library but now i have 3 errors.
Strange. Before update had no errors in compilation

Timon2, 2016-10-27:
all is ok, thx microbuilder

microbuilder, 2016-10-27:
Please file this as an issue on Adafruit's forums. It's inappropriate here to hijack an unrelated issue, thanks.

cmaglie, 2016-11-02:
Done, the library will disappear at the next weekly cleanup round.

closed 2016-11-02 by cmaglie