arduino/Arduino #5531: New Library: ToneLibrary

daniel-centore, 2016-10-28:
URL: https://github.com/daniel-centore/arduino-tone-library

This is a fixed fork of [an abandoned Google Code project](https://code.google.com/archive/p/rogue-code/wikis/ToneLibraryDocumentation.wiki).

shiftleftplusone, 2016-10-31:
hello,
does your lib work for the Arduino Due (ARM Cortex M3), too?

daniel-centore, 2016-10-31:
Don't have a Due to test with and I'm honestly not sure.

cmaglie, 2016-11-02:
I've added the library.

BTW your library.properties shows version 1.7 on the 1.7.1 tag:

https://github.com/daniel-centore/arduino-tone-library/blob/v1.7.1/library.properties

please fix that:

1) remove the 1.7.1 tag
2) update library.properties
3) commit and push
4) re-add the new tag 1.7.1

once done, the change will be picked up by our server within an hour and the release 1.7.1 will be published automatically, no need to further comment here.

closed 2016-11-02 by cmaglie