arduino/Arduino #5534: Arduino IDE: Missing a way to edit files in src folder

rkost, 2016-10-30:
So you decided to only compile files that are in the src folder of a sketch (That was why #5296 was created). This is totally ok but for it broke the IDE for me. Before the IDE automatically opened all files (cpp and h) that were in the same folder as the ino file was. Now it only opens the ino file.

Personally I only use the arduino.exe in the command line and have a different editor to write code but when I give my code to arduino IDE users it is not really possible to explain how they can open files like 'SketchDir'/src/example[.h/.cpp]. Is there a workaround or what am I missing? 

By the way: When I use the menu item "sketch: Add File..." the file(s) are added to the main directory of the sketch so it won't compile if the files are .h and .cpp and included in the main sketch. *Confused*

matthijskooijman, 2016-10-30:
> So you decided to only compile files that are in the src folder of a sketch 

That's not true (or should not be anyway). Files inside the main directory are compiled, _as well as_ files in (subdirectories of) the `src/` directory. So most users can just stick to files in the main directory, next to the .ino file, which will be opened in the IDE and compiled. For more complex projects that need more files and more structure, the `src/` directory can be used to divide files into subdirectories. It is indeed inconvenient that these files can not currently be edited with the IDE (and support for this should be added in the future), but I suspect that most users of the `src/` folder will be using an external editor anyway.

However, reading your report closely, you say that `.cpp` files that are adjacent to the `.ino` file, so in the root of the sketch directory, are not compiled. What version are you using exactly? I'm pretty sure it works for me (though I have a version from git from a while back, not a released version). If it does not work, that's a bug that should be diagnosed and fixed.

rkost, 2016-10-31:
Okay. Sooo. I got it. And it seems that i missunderstood what you told me in #5296 (I was thinking that I have to move everything into src folder except from the ino file).
The thing why my code did not compiled with 1.6.10 and higher was that I had the following directorys with code in it (mostly cpp classes in .h and .cpp):

```
project/project.ino
project/someclass.h
project/someclass.cpp
project/internals/someinternalclass.h
project/internals/someinternalclass.cpp
```
(Simple Example included in attachment). Now in #5296 I was told that files in subdirectorys not included in src are not compiled anymore in newer versions so I refactored:
```
project/project.ino
project/src/someclass.h
project/src/someclass.cpp
project/src/internals/someinternalclass.h
project/src/internals/someinternalclass.cpp
```

But like @matthijskooijman mentioned it also works when I put all files that are not inside subdirectories directly into the project directory (like this):
```
project/project.ino
project/someclass.h
project/someclass.cpp
project/src/internals/someinternalclass.h
project/src/internals/someinternalclass.cpp
```
However the first version is not compiling anymore but that is fine for my so far. It would be nice if it would be possible to edit files in src in the IDE but it is not that important to me. 
Thank you for your explanation @matthijskooijman !

closed 2016-10-31 by rkost

matthijskooijman, 2016-10-31:
@rkost, thanks for your extensive followup, good to hear things are cleared up :-)