arduino/Arduino #5539: link time optimization (-lto) of AVR breaks inclusion of debug info in elf file

WestfW, 2016-10-31:
In recent versions of avr-gcc, the -g option is needed during compile in order to get enough debug information into the final elf file for things like a combined source/disassemebly (avr-objdump -SC foo.elf) to work.   Now that link time optimization has been implemented, the "link" command will need the -g option as well as the individual compiles...

(It's possible that some other compile-time options also need to be moved to the link statement as well, but "-g" is the one I noticed.)

facchinm, 2016-11-02:
Hi @WestfW ,
does it impact the size of the final hex (once the debug sections have been removed of course)? If it doesn't, I believe we could simply add the `-g` flag everywhere. Otherwise we'll need to provide a way to override the commandline in order to produce a debuggable elf

matthijskooijman, 2016-11-03:
The .hex file is unaffected, so I would encourage adding -g.

commit 35e45c9f (2016-12-15): [AVR] Add -g to elf.flags to allow inclusion of debug symbols in elf

Fixes #5539
closed 2016-12-15 by cmaglie