arduino/Arduino #5553: Class templates have probelsm when the body of its functions are in a seperate file from its declaration

Derks2, 2016-11-02:
If you create a class template where the declaration is in the .h file and the bodies for the functions are in a .cpp files, the code will compile as long as the user does not call any of the classes functions. As soon as the user add a call to any of the functions in the class the compiler complains it can not find the function.
Now if you take those same function, with out changing them, and put them at the end of the .h file with the lass declaration, then compiler doesn't have any problems and compiles.

cmaglie, 2016-11-02:
It's how templates are supposed to work, or maybe I'm missing something?

Chris--A, 2016-11-02:
@cmaglie correct.

@Derks2 If you want to put the implementation in a cpp file, you will need to instantiate the particular variations in the cpp file also. The draw back is, users can only create the types you have already instantiated, any others will cause the same error you are getting now. This is fine if your template is quite specific and will break/work improperly with generic use.

matthijskooijman, 2016-11-03:
Yup, this is how C++ templates work, check out docs about C++ to find out more (and as @Chris--A points out, you'll need explicit template instantiations to fix this, or just put everything in the .h file). I'm closing this issue, as this is not a bug or something Arduino can/should fix.

closed 2016-11-03 by matthijskooijman