arduino/Arduino #5554: Selecting serial port of programmer

petersheeren, 2016-11-02:
We're adding an AVR in-system programmer to the Arduino IDE. This programmer comes with its own software i.e. it uses a dedicated program instead of avrdude. At this point the programmer is fully functional from within Arduino IDE.

The programmer exposes a USB-to-serial interface similar to many Arduino boards. This means the system dynamically assigns a COMx (Windows) or /dev/ttyUSBx (Linux) path to the programmer the same way it does for Arduino boards. The IDE shows the programmer's serial port in menu item Tools->Port.

We found that Arduino IDE doesn't offer a mechanism for selecting a serial port for the programmer. As a workaround we're using the selection from menu item Tools->Port which is known as {serial.port} in platform.txt. This works, but doing so has a negative impact on the user-experience as the user has to select the correct serial port each time he or she uploads a sketch and then opens the serial monitor.

I'd like to request the addition of a new menu item in menu Tools that allows the user to select the serial port of a programmer featuring a serial interface.

matthijskooijman, 2016-11-03:
> As a workaround we're using the selection from menu item Tools->Port which is known as {serial.port} in platform.txt

I don't think this is a workaround, but intended behaviour right now. I.e. it's already used like this by builtin programmers: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/programmers.txt#L6

This is somewhat related to #2073 as well.

petersheeren, 2016-11-13:
I see. Ofcourse there's really no other choice whether one calls it a workaround or intended behavior, that's why built-in programmers use Tools->Port.

Anyway, in the meanwhile we'll advise users to use an external serial monitor for their boards (e.g. PuTTY) when using our programmer so they don't have to switch ports in Tools->Port all the time.

per1234, 2022-12-01:
Now tracked at https://github.com/arduino/arduino-ide/issues/583

closed 2022-12-01 by per1234