arduino/Arduino #5555: Addition of MIDI library

Franky47, 2016-11-02:
Hello,

I recently refactored my [MIDI Library](https://github.com/FortySevenEffects/arduino_midi_library) for 1.5 rev2.1 format compatibility, and I've just released its version 4.3, available here:
https://github.com/FortySevenEffects/arduino_midi_library/releases/latest

If you could add it to the registry for the Arduino Library Manager, that would be nice.

Thanks !

shiftleftplusone, 2016-11-03:
hello,
is this MIDI lib also compatibel to Arduino ZERO, M0,  and DUE boards?

Franky47, 2016-11-03:
Yes. All boards/platforms that expose a Serial interface (HardwareSerial or SoftwareSerial) are compatible.

facchinm, 2016-11-04:
Hi @Franky47 ,
we can't actually import the library since the version number in `library.properties` is still 4.2 while the tag is 4.3. Can you fix it? Thx :slightly_smiling_face:

Franky47, 2016-11-04:
Ah, I must have missed that bugger, I'll fix that rightaway.

Franky47, 2016-11-04:
It should be all good now:
https://github.com/FortySevenEffects/arduino_midi_library/tree/4.3.1

facchinm, 2016-11-04:
Added, thank you!

closed 2016-11-04 by facchinm