arduino/Arduino #5558: Wrong calculation of CRC?

leomil72, 2016-11-03:
It's 2 days that I'm trying to have the Arduino IDE recognize my new Tiny core v.1.0.6. I've updated the JSON file and the IDE sees the new release but when I come to update it, I get "CRC doesn't match. File is corrupted".
I'm very surprised because I calculated the SHA-256 sum using 3 different tools (shasum on OSX, shasum on Linux and an online tool) and I got the same CRC from the 3 tools. So it should be correct. 

The JSON file is this one:
http://www.leonardomiliani.com/repository/package_leonardomiliani.com_index.json

You can directly grab the Tiny core from here:
https://github.com/leomil72/tiny/releases/download/1.0.6/tiny-1.0.6.zip

The SHA-256 sum is:
d9ec07e77c1f11434e3ffbadd5520dd2daf7b4e1b3d9396014022f5bb0f19ed8

Why the IDE continue to say that the CRC is wrong?

per1234, 2016-11-04:
It works for me. I suspect you are encountering https://github.com/arduino/Arduino/issues/5394, which has been fixed in the hourly build. You can also solve it by deleting the package file from Arduino15/staging/packages.

leomil72, 2016-11-05:
Thank you per1234, you centered the issue. I compiled the IDE from the sources and the error was fixed. Thank you, I close here.

closed 2016-11-05 by leomil72