arduino/Arduino #5560: Typos in Secrets of Arduino PWM

greenonline, 2016-11-03:
In https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM, there are two typos:

1. Spelling of "frequency" - "Note that both outputs have the same frequncy, matching the frequency of a complete timer cycle."

2. Missing header. The heading tag of "Off-by-one" has been lost and as such the heading appears as part of the text. As this section is referred to earlier on in the text (in the section **Phase-Correct PWM example**: "See the explanation below under "Off-by-one" "), this issue makes it a bit difficult to find the section referred to. The issue is here, in this quoted text: "... see below for details. Off-by-one You may have noticed that fast PWM and phase-correct PWM seem to be off-by-one with respect to each other, "

So "Off-by-one" should be a header in orange text, and "You may have noticed..." should start on a new line.

greenonline, 2016-11-04:
I have editted the original bug report and added the following:

There are a number of other issues with this document. If you compare it with the original blog on Ken Shirriff's website, http://www.righto.com/2009/07/secrets-of-arduino-pwm.html, you can see additional issues.

Issues 3 - 7, are missing graphs that are referred to in the text. There absence makes it difficult to understand the reasoning of the text.

3. The graph is missing in the "Fast PWM" section
4. The graph is missing in the "Phase-Correct PWM" section
5. The graph is missing in the "Varying the timer top limit: fast PWM" section
6. The graph is missing in the "Varying the timer top limit: phase-correct PWM" section
7. The graph is missing in the "Varying the timer top limit: phase-correct PWM" section.
8. As with point 2, the heading of the section "Timers and the Arduino" has become part of the text
9. As with point 2, the heading of the section "Troubleshooting" has become part of the text

akash73, 2017-02-10:
thanks @greenonline for point it out.
we are working on it

jschrempp, 2017-02-11:
In addition to the issues noted above:

8. Under the current heading "The Atmega 168/328 timers." The heading of the section "Timer Registers" has become part of the text.

9. The second paragraph calls PWM output "a digital square wave." My impression is that a square wave always has a 50% duty cycle. The PWM output might better be called "a digital rectangular wave." Just a suggestion.

akash73, 2017-02-13:
@greenonline @jschrempp 
page completely reworked and published.

thanks so much for your advice.

greenonline, 2018-06-21:
@akash73 @jschrempp 
Hi, I have just revisited [the page](https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM) again, and it still looks exactly the same, with the same issues. Was the reworked page published to the same location?

akash73, 2018-06-22:
@SimonePDA please check.

per1234, 2022-07-17:
Thanks all for your attention to the quality of the Arduino documentation. I have reviewed the page and found that all the reported issues have been resolved now.

closed 2022-07-17 by per1234