arduino/Arduino #5564: `undefined reference` occurs in building SchedulerARMAVR

asukiaaa, 2016-11-05:
I sometime use a library `SchedulerARMAVR` to create a multi tasking project.

https://github.com/fabriceo/SCoop/tree/master/SchedulerARMAVR

It succeeded building with using 1.6.8 and 1.6.9 but failed with using 1.6.10, 1.6.11 and 1.6.12.

IDE outputs the following error.

```
Arduino: 1.6.12 (Linux), Board: "Arduino Nano, ATmega328"

/tmp/ccfwLs4I.ltrans0.ltrans.o: In function `coopTaskStart':
ccfwLs4I.ltrans0.o:(.text+0x8): undefined reference to `coopSchedule'
/tmp/ccfwLs4I.ltrans0.ltrans.o: In function `coopDoYield':
ccfwLs4I.ltrans0.o:(.text+0x30e): undefined reference to `coopSchedule'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

The error occurs even though I remove `.arduino15`.

I use Ubuntu16.04 64bit.
Is there any way to avoid this error?

Thanks.

per1234, 2016-11-05:
I was able to fix the issue by making the following change:
https://github.com/fabriceo/SCoop/pull/3

asukiaaa, 2016-11-05:
@per1234 
Thank you!
The change fixed this issue on 1.6.12 and my program works.

I wait merging your pull request or create my repository.

closed 2016-11-05 by asukiaaa