arduino/Arduino #5570: Update Library Request

closedcube, 2016-11-06:
Ciao,

Please add a new version for the following libraries:
https://github.com/closedcube/ClosedCube_OPT3001_Arduino
https://github.com/closedcube/ClosedCube_OPT3002_Arduino

Many thanks

cmaglie, 2016-11-09:
About OPT3001, you already have: https://github.com/closedcube/ClosedCube_OPT3001_Library.git
shall we remove that one before adding the new one?

cmaglie, 2016-11-09:
About OPT3002, the new release is already published: once the library is added to the library manager you can autonomously publish a new release by adding a new tag, our crawler checks every hour for a new release, no need to request again here.

closedcube, 2016-11-10:
Yes, please "old" version of OPT3001 - https://github.com/closedcube/ClosedCube_OPT3001_Library.git

re: OPT3002 - thanks for the details

cmaglie, 2016-11-18:
Done!

closed 2016-11-18 by cmaglie