arduino/Arduino #5574: new feat.: seperate Find (^f) and replace (^h) functions, + search only whole words !

shiftleftplusone, 2016-11-07:
hello,
this is a feature request,
there are still some features missing with respect to search and replace funtions:
1.) seperate Find (^f) and replace (^h) functions would be needed 
(no "replace" when just "find" had to be used)
2,) an additional option (check box) for search only for "whole words" is needed

cousteaulecommandant, 2016-11-19:
Or at least add the possibility to use a different editor (make Arduino IDE detect when the .ino file has been changed externally and update it; many IDEs do this).  I was going to suggest using Geany or another editor with this feature, but that's not quite convenient if you have to close and reopen the project every time you edit it externally.

matthijskooijman, 2016-11-19:
> Or at least add the possibility to use a different editor 

This is possible by setting the "use external editor" preference, which disables edits in the IDE and reloads the files whenever you verify or compile.

shiftleftplusone, 2016-11-20:
no, my request was just about the original Arduino IDE as  new "find" and "search/replace" feature requests. 
Please keep on-topic, thank you.

per1234, 2017-07-06:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3727.

closed 2017-07-06 by per1234