arduino/Arduino #5576: Leonardo (tinkerkit)

BallscrewBob, 2016-11-07:
OS Win 7 x64 and either CREATE or IDE 1.6.12.
Stalls or fails on uploading to Leonardo.

Not confined to the tinker-kit as also mentioned in Arduino forum.

Output from CREATE first followed by IDE.

`arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/299995156 -hardware arduino-builder/hardware -hardware arduino-builder/packages/cores -tools arduino-builder/tools -tools arduino-builder/packages/tools -built-in-libraries arduino-builder/latest -libraries /tmp/591972169/pinned -libraries /tmp/591972169/custom -fqbn arduino:avr:leonardo -verbose=true /tmp/591972169/BlinkWithoutDelay

Using board 'leonardo' from platform in folder: /home/admin/builder/arduino-builder/packages/cores/arduino/avr

Using core 'arduino' from platform in folder: /home/admin/builder/arduino-builder/packages/cores/arduino/avr

WARNING: Category '' in library ArduinoUnit is not valid. Setting to 'Uncategorized'

WARNING: Category '' in library Parse Arduino SDK is not valid. Setting to 'Uncategorized'

WARNING: Category '' in library Pushetta is not valid. Setting to 'Uncategorized'

WARNING: Category '' in library SparkFun MG2639 CellShield is not valid. Setting to 'Uncategorized'

WARNING: Category '' in library StereoSID is not valid. Setting to 'Uncategorized'

Detecting libraries used...

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/tmp/299995156/sketch/BlinkWithoutDelay.ino.cpp" -o "/dev/null"

Generating function prototypes...

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/tmp/299995156/sketch/BlinkWithoutDelay.ino.cpp" -o "/tmp/299995156/preproc/ctags_target_for_gcc_minus_e.cpp"

"/home/admin/builder/arduino-builder/tools/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/299995156/preproc/ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/tmp/299995156/sketch/BlinkWithoutDelay.ino.cpp" -o "/tmp/299995156/sketch/BlinkWithoutDelay.ino.cpp.o"

Compiling libraries...

Compiling core...

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino/wiring_pulse.S" -o "/tmp/299995156/core/wiring_pulse.S.o"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino/WInterrupts.c" -o "/tmp/299995156/core/WInterrupts.c.o"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino/hooks.c" -o "/tmp/299995156/core/hooks.c.o"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino/wiring.c" -o "/tmp/299995156/core/wiring.c.o"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino/wiring_analog.c" -o "/tmp/299995156/core/wiring_analog.c.o"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino/wiring_digital.c" -o "/tmp/299995156/core/wiring_digital.c.o"

"/home/admin/builder/arduino-builder/packages/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10611 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino" "-I/home/admin/builder/arduino-builder/packages/cores/arduino/avr/variants/leonardo" "/home/admin/builder/arduino-builder/packages/cores/arduino/avr/cores/arduino/wiring_pulse.c" -o "/tmp/299995156/core/wiring_pulse.c.o"

[... cut ...]
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

facchinm, 2016-11-07:
Well, it looks like the port is not being touched (by Create) while it's being touched by the Java IDE but the bootloader is not being recognized.
Does programming the board works without the Tinkerkit shield?

BallscrewBob, 2016-11-07:
It isn't a shield per-se.
[http://www.carobot.cc/how-to/tinkerkit-tutorials/tinkerkit-tutorial-lcd-06-twi/](url)

The only thing that could be removed is the LCD itself.
It worked fine on both CREATE and the IDE about a week ago.
There are also a couple of mentions of LEO issues in the forum recently hence my supposition that its builder related or something that has changed in that area.

Am going to try an earlier IDE shortly from an unzipped locations and will report back.

BallscrewBob, 2016-11-07:
Odd the link now appears pooched..

Plain txt version

http://www.carobot.cc/how-to/tinkerkit-tutorials/tinkerkit-tutorial-lcd-06-twi/

facchinm, 2016-11-08:
There has been no relevant change in the IDE in the last week, could you confirm that it works with 1.6.12 or older versions? Thx

BallscrewBob, 2016-11-08:
Works in 1.6.10 from an unzipped copy.
Still fails in CREATE.
1.6.12 worked this morning...

facchinm, 2016-11-08:
Ok, the problem is in Create. Closing this one, keeping track in Create forum section and create repo. Thx

closed 2016-11-08 by facchinm