arduino/Arduino #5577: IDE 1.6 hampers serial speed on Due (SAM3X8E)

cjbaar, 2016-11-07:
I have been using several Arduino Due boards under the beta IDE (1.5.8). Previously, I was able to successfully maintain a continuous serial connection between two boards (using native Serial ports) at 5,000,000bps. Using the exact same boards and code, but being programmed using a current IDE (1.6.11), I am not able to get above roughly 500,000. This is only 10% of the previous speed. I am not sure what may have changed in the Serial libraries from beta, but I would like request it be reverted to allow higher speeds.

dlabun, 2016-11-07:
There's a pretty big jump between 1.5.8 and 1.6.11... Did you try another 1.6 releases?

cjbaar, 2016-11-07:
No; this is the first version of 1.6 that I have tried using for programming SAM.

facchinm, 2016-11-08:
The issue you are encountering looks like a side effect of some irq-related patch. This one https://github.com/arduino/Arduino/commit/2fedb00552e45aaca1623789138d223690244f43 may be relevant. Anyway, if the previous code was not IRQ driven, it was probably faster due to the lack of context switch but less user friendly (since you have to poll continuously the registers to check if some data has arrived). Moving to DMA based transfer would solve both the issue but it's a **huge** task

cjbaar, 2016-11-15:
Thanks, facchinm. I've written a few low-level libraries, but your comments are bit beyond my level of experience.  :)  Also, thanks for adding the SAM label; I couldn't figure out how to do that.

In this particular case, I am using one Due simply as a repeater, relaying data from the first Due, refactoring it, and sending back out via some custom bit-banging. I am sending "frames" of 480 bytes, and averaging about 70-80 frames/second, so the Serial ports are fairly active. I am not sure how the interrupts come into play, but I guess for now I will be keeping the 1.5.8 beta version around. Also, if it makes a difference, I am running the IDE on OSX.

closed 2017-02-08 by agdl