arduino/Arduino #5581: Arduino IDE on Secondary monitor issue

mefi01, 2016-11-08:
Hello.

I experience a bug if I use the Arduino IDE with multiple screens.
I use Arduino 1.6.8 on Windows 10 Pro.

First, drop the IDE window on the secondary Screen and close it. After unplug the screen and try to reopen the program. The loading window shows on the main screen, but when it finish, you can't see. 
The program will be on the secondary monitor , but it doesn't connected!

The Serial Monitor is affected too.

It's very annoying, mostly when i forgot to close the IDE or the Serial Monitor on the laptop's screen and I don't have any monitors with me.

Do you also have such an error?

septillion-git, 2017-02-01:
Argg, hitting this same wall. Apparently the last time I used the Serial Monitor I didn't do that on my primary screen. And now I'm connected via remote desktop (so only one screen as far as Windows is concerned) and I can't get the Serial Monitor to show :/

Arduino IDE 1.8.1 on Windows 10

PaulStoffregen, 2017-02-01:
To recover, you're probably going to have to edit your preferences.txt file.  It's in a hidden folder.  Quit Arduino, so your changes don't get overwritten.  Then in Windows Explorer, type %LocalAppData% in the location, then look for the preferences.txt file in the Arduino15 folder.  Simply deleting the lines about the serial monitor position should be enough.

septillion-git, 2017-02-01:
Thanks for the tip! I used the work around of starting a portable version of the IDE 0:-) When I'm back home it's not a problem anymore when it pops to the second monitor.

But it does show the bug and the need for the IDE to check if the location still exists. Probably the same for the height and width...

twanek, 2017-08-05:
hello!

i see this bug was first reported back in 2016. but in 2017, with latest ide (1.8.3) it is still the same (tested on windows 7 and windows 10)!
please resolve this issue, it is VERY annoying! lots of times i simply can't "access" my serial monitor...

thanks!

facchinm, 2017-08-07:
Digging a bit in this problem, it looks like:
- there is a lot of "convoluted" logic to handle the restore (https://github.com/arduino/Arduino/blob/master/app/src/processing/app/AbstractMonitor.java#L61)
- `last.screen.height` and `width` should be updated on IDE close. Instead, they are saved in `storeSketches()` , which is also called when you open a sketch (see https://github.com/arduino/Arduino/blob/master/app/src/processing/app/Base.java#L839)

So on IDE startup `if ((screen.width == screenW) && (screen.height == screenH))` is always true because the value is getting overridden by any sketch being reopened. 

I'm preparing a PR to always avoid the screen size check and only trust the dimensions (if it fits restore the location, otherwise start as default). Can anyone spot any problem in this approach?

twanek, 2017-08-07:
what is the difference between screen size and dimension?

facchinm, 2017-08-07:
@twanek I was referring about the serial monitor dimensions :slightly_smiling_face: Could you test https://github.com/arduino/Arduino/pull/6589#issuecomment-320611971 and report if the problem is solved (no more disappearing serial monitor)? Thanks!

twanek, 2017-08-07:
yes, it is working now, in my setup they are different, the laptop is 17.5" and the external is 22".
but what if the 2 displays have the same resolution / dimension?

facchinm, 2017-08-07:
It shouldn't matter, since the "virtual screen" when they are both connected will be (roughly) (screen1.width + screen2.width) x (max(screen1.height,screen2.height)) (if they are arranged side by side). This is the best case and the serial monitor will always fit in these dimensions. If you move anything and the window doesn't fit anymore if should be repainted to 0,0 with safe dimensions.

twanek, 2017-08-07:
ah, ok, i understand now. didn't know how the "virtual screen" works in the background.
i hope this solution for the bug will be implemented in the official ide version soon.

thank you for the help!
have a nice day,

closed 2017-08-18 by cmaglie

camilov87, 2019-09-08:
Going to C:\Users\user\AppData\Local\ and deleting the Arduino15 folder solved the problem for me.

I was using dual monitor

celiagarridoh, 2019-11-02:
Hi, on Windows 10 this worked for me: pressing **Windows+Shift+up** or **Windows+Shift+down**.

Regards,
Celia