arduino/Arduino #5586: [website] "Authentication Required" on product pages

per1234, 2016-11-10:
The following product pages linked from the main chart on https://www.arduino.cc/en/Main/Products cause an "Authentication Required" dialog to appear:
[image]
https://www.arduino.cc/en/Main/ArduinoBoardMini
https://www.arduino.cc/en/Main/Robot
https://www.arduino.cc/en/Main/GTFT
https://www.arduino.cc/en/Main/ArduinoWiFiShield
https://www.arduino.cc/en/Main/ArduinoWirelessProtoShield

akash73, 2016-11-11:
@per1234 thanks for noticing.
pages updated, now they don't show anymore the auth dialog box.

oqibidipo, 2016-11-11:
But wait, there's more... 

There are still links to edit.arduino.cc at least on the following pages: 

https://www.arduino.cc/en/Main/ArduinoBoardEsplora
https://www.arduino.cc/en/Main/ArduinoBoardEthernet
https://www.arduino.cc/en/Main/ArduinoBoardLeonardo
https://www.arduino.cc/en/Main/ArduinoBoardMegaADK
https://www.arduino.cc/en/Main/ArduinoEthernetShieldV2
https://www.arduino.cc/en/Main/ArduinoMotorShieldR3
https://www.arduino.cc/en/Main/ArduinoUSBHostShield
https://www.arduino.cc/en/Main/ArduinoWirelessShield
https://www.arduino.cc/en/Main/MiniUSB
https://www.arduino.cc/en/Main/USBSerial

Details (grep output): 
[edit_arduino_cc.txt](https://github.com/arduino/Arduino/files/586565/edit_arduino_cc.txt)

akash73, 2016-11-15:
Thanks also to @oqibidipo ! we have fixed the broken / uncorrect links.
I'll leave this issue open til some days, if you notice some other glitches.

per1234, 2017-01-17:
I haven't found any more links to edit.arduino.cc since and all the pages listed above are fixed so I'm going to close this issue. Thanks akash73!

closed 2017-01-17 by per1234