arduino/Arduino #5587: Problem changing name to library

alejandrorosas, 2016-11-10:
[image]

There's an issue on the first version name, and on the last version I've changed "Driver" to "Drive" and now when I install the new version at the library manager there are two libraries.

Could you remove the first version? Or is there anything I can do?

Thanks!!

agdl, 2016-11-11:
Can you please specify the github link?

alejandrorosas, 2016-11-11:
Yes! Here: https://github.com/IngeniaMC/Ingenia-Serial-Servo-Drive-Library

alejandrorosas, 2016-11-16:
@agdl Is there anything I can do to help or fix the issue?

cmaglie, 2016-11-18:
It would have automatically disappeared this weekend, during the weekly clean-up, BTW I forced a manual clean up to speed up things, and now it's ok!

closed 2016-11-18 by cmaglie