arduino/Arduino #5590: Arduino IDE can't access /dev/ttyUSB0 unless soft link created!

githubdispostable, 2016-11-11:
Hello,

I've installed the arduino IDE on Debian 7. When running the application as sudo, selecting the correct board from tools (Nano atmega328) and selecting the serial port for my board (/dev/ttyUSB0) I get an error when trying to use the serial console or upload sketch.

The error I get is http://pastebin.com/raw/e3Bs5Jbw

To fix this error I did "rm -rf /dev/ttyS0", then "ln -s /dev/ttyUSB0 /dev/ttyS0".

Thanks

cousteaulecommandant, 2016-11-19:
Not an answer to your problem, but rather than using `sudo` for everything I'd add yourself to the `dialout` group so that you can use `/dev/ttyUSB0` (and `/dev/ttyACM0`) as a regular user.
`sudo usermod -a -G dialout $USER`

Running arduino as `sudo` might be "irreversible" though, depending on how you used it (i.e. you might always need `sudo` from now on because it'll have changed the permissions of the config file).  If this is the case, you can fix this with
`sudo chown -R $USER:$USER ~/Arduino ~/.arduino15`
(If instead of `sudo` you use `gksudo` or `sudo -H` then Arduino will mess with root's config files rather than your user's.)

facchinm, 2016-11-21:
Hi @githubdispostable , 
as @cousteaulecommandant suggested it's better not to launch the IDE as root. 
Your problem is probably related with the serial-to-usb converter you are using and/or the native serial port (`ttyS0`). If the problem persists or you want to find a proper solution, the forum is the right place to ask for help. 
Thanks

closed 2016-11-21 by facchinm