arduino/Arduino #5597: Plz add 2 more libraries: Adafruit_MAX31865 & Adafruit_MAX31856

ladyada, 2016-11-15:
https://github.com/adafruit/Adafruit_MAX31865
and
https://github.com/adafruit/Adafruit_MAX31856
(similar names but different!)

cmaglie, 2016-11-18:
https://github.com/adafruit/Adafruit_MAX31865, done!

https://github.com/adafruit/Adafruit_MAX31856 is missing a `library.properties` file!

ladyada, 2016-11-18:
*sigh* - its been a week! ok fixed now? :)

cmaglie, 2016-11-18:
not yet ;-)... the tag is still pointing to the old version (without the file)

https://github.com/adafruit/Adafruit_MAX31856/tree/1.0.0

just remove and make the tag again and we are done,
BTW the library is already listed in the server, once the tag is fixed it will be picked up automatically by the server (within an hour).

ladyada, 2016-11-18:
to avoid making yet a bigger mess of things, just bumped version and re-released 1.0.1 :)

closed 2016-11-18 by ladyada